WEC-Sim/WEC-Sim #850: [Theory or Implementation] BEMIO normalization error using Capytaine output

Dadidal, 2022-04-19:
Hello! I am trying to create .h5 file using BEMIO based on results from BEM code Capytaine. When I run the bemio_Capytaine.m file in MATLAB the following error is occured:

> Error using  / 
Integers can only be combined with integers of the same class, or scalar doubles.
Error in Normalize (line 49)
    hydro(F).C = hydro(F).C/(hydro(F).g*hydro(F).rho);
Error in Read_CAPYTAINE (line 403)
hydro = Normalize(hydro);  % Normalize the data according the WAMIT convention
Error in bemio_Capytaine (line 3)
hydro = Read_CAPYTAINE(hydro,'.\ship_Capytaine.nc');

I am acting according to following algorithm:
1. Create .stl geometry model using CAD software
2. Calculation in Capytaine:
- import geometry
- calculate hydrostatics
- write KH.dat and Hydrostatics.dat files to results folder
- solve problems
- write Capytaine database to ship_Capytaine.nc file
3. Copy KH.dat, Hydrostatics.dat, ship_Capytaine.nc files to 'Wec-SIM project folder\hydroData\
4. Run bemio_Capytaine.m

It seems like normalization of Capytaine output doesn't work properly. Could you plese help me to solve this problem?
I attached the archive with Capytaine and Wec-SIM working folders and files.
[Working folders.zip](https://github.com/WEC-Sim/WEC-Sim/files/8510708/Working.folders.zip)

akeeste, 2022-04-19:
Hi @Dadidal 

Thank you for opening an issue. Please maintain the template structure as it helps us ensure that all required information is present. Can you add your workstation information to this issue?

OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
[MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [e.g. 2019b, 2020a,]
[WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [e.g. v4.0, v4.1]

I will look at your BEMIO files and see what might be causing this error.

Thanks,
Adam

Dadidal, 2022-04-20:
@akeeste 
Thank you for help!
> Can you add your workstation information to this issue?
OS: Windows 10 x64
MATLAB/Simulink Release: R2021a
WEC-Sim Release: v4.4

akeeste, 2022-04-20:
@Dadidal 

This error is caused because the density is being written/read as an integer. Line 39 in Normalize.m: 
``hydro(F).C = hydro(F).C/(hydro(F).g*hydro(F).rho);`` fails because a MATLAB double (C) is being divided by an integer (rho).

To immediately fix this problem, you can change your density definition in ``AE.py`` to be a float by adding a decimal:
``rho_water = 1000.  # Water density in kg/m3``

I will investigate ways to systematically catch these type mismatches in BEMIO. Please let me know if you have any additional questions on this issue.

Adam

Dadidal, 2022-04-21:
@akeeste 

I changed the rho_water to integer and now it works.
Thank you a lot!

closed 2022-04-21 by Dadidal

davidfdomingos, 2023-02-21:
> @Dadidal
>
> This error is caused because the density is being written/read as an integer. Line 39 in Normalize.m: `hydro(F).C = hydro(F).C/(hydro(F).g*hydro(F).rho);` fails because a MATLAB double (C) [...]

I ran into the same issue, so I just hydro(F).C = hydro(F).C/double(hydro(F).g*hydro(F).rho) in the entire  normalizeBEM.m and it runs smooth now.