WEC-Sim/WEC-Sim #851: [WEC-Sim Applications] How to calculate the power extracted without a controller in WECCCOMP model?

TamerAbdelmigid, 2022-04-20:
Question: How to calculate the power extracted without a controller in WECCCOMP model?

I am trying to get a base line to see how much energy will this model extract without control, from what I understand `power = -output.ptos.forceActuation(:,3) * output.ptos.velocity(:,3)`
and inside the linear motor block, cmd_force = forceActuation. So, if I set the actuation force to a constant of 0, there will be no power produced. Hence, how can I calculate the power in the absence of a controller?

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - MATLAB 2020b
 - WEC-Sim 4.4

nathanmtom, 2022-04-21:
@TamerAbdelmigid You are correct that the base WECCCOMP model does not have an default controller options that generate an example energy extraction value or performance. 

The simplest way to implement a passive linear damper is to select a `cc` variable in the `controller_init.m` function available in the WECCCOMP repository. The `cc` variable will add a resistive damping value to response of the PTO but will need to be tuned to optimize power extraction based on the sea state of interest. 

A second method is that in our `dev` branch for our WEC-Sim applications repository, https://github.com/WEC-Sim/WEC-Sim_Applications/tree/dev, there is a `WECCCOMP` folder where we have other examples with tuned controllers for this WEC-Sim application. See https://github.com/WEC-Sim/WEC-Sim_Applications/tree/dev/WECCCOMP/WECCCOMP_Nonlinear_Model_Predictive.

Please let me know if this has answered your inquiry. 

Cheers, 
Nathan

TamerAbdelmigid, 2022-04-21:
@nathanmtom Thank you so much , this provide a baseline controller to test against as well as a state-of-the-art one.
I will chose the proportional controller with proportional gain equal to zero as a no-controller case.

Unfortunately, I tried to install WECSim 5.0 dev to run the example and got faced with `"Unrecognized function or variable 'Read_WAMIT'"`
So, sorry for the trivial question, but how to properly run the NMPC WECCCOMP example?

nathanmtom, 2022-04-21:
@TamerAbdelmigid I am glad to hear that you this application case may help you in your own work. 

To clarify, if you set the proportional gain equal to zero the model will not extract any power. This does not seem to be consistent with your initial inquiry about obtaining a baseline power estimate.

The WEC-Sim development team has been completing a restructuring of our variable labeling, see [here ](https://github.com/WEC-Sim/WEC-Sim/pull/840), so you may need to rename some of the function names such as `Read_WAMIT` to `readWAMIT` and other functions within the BEMIO script. 

The WEC-Sim team will be releasing our v5.0 in the near future which should resolve the naming issue, but cannot commit to when this will be completed.

TamerAbdelmigid, 2022-04-21:
@nathanmtom , Thanks again for the clarification for both issues, I will correct the naming as provided, and for the damping coefficient I will set it to a value tuned for the sea state of interest as you mentioned earlier.

Appreciate your help.

closed 2022-04-23 by TamerAbdelmigid