WEC-Sim/WEC-Sim #857: Running WEC array

SudharsanKalidoss, 2022-04-26:
Hi all,

    I'm trying to run a WEC farm with two WECs (similar to the RM3 WEC system). When running WEC-Sim for a single WEC system, my simulation ran fine and I got the required results. For an array system, I'm getting an error (attached below). The array system consists of two WEC systems and a stationary tower. The WEC1 is placed at (0,0,0), the tower is at (0,96,0), and the WEC2 is placed at (0,192,0). I have attached my WEC-Sim input files. On checking the closed issues, I found a similar error (#549) was answered. The WEC-Sim has pointed out that the error occurred because of the version number. Also, it is reported that the error is rectified. I'm currently using MATLAB R2021b, MATLAB version 9.11, and SIMULINK version 10.4. I using WEC-Sim version 4.4.

Thanks
[image]
[SRPA_farm.zip](https://github.com/WEC-Sim/WEC-Sim/files/8564465/SRPA_farm.zip)
[SRPA_farm_hydro.zip](https://github.com/WEC-Sim/WEC-Sim/files/8564473/SRPA_farm_hydro.zip)

SudharsanKalidoss, 2022-04-28:
Hi all,

    I found the error. I didn't number rigid bodies properly. It is working fine now.

Thanks

closed 2022-04-28 by SudharsanKalidoss

akeeste, 2022-05-02:
Hi @SudharsanKalidoss 

Thank you for the update and I'm glad your issue was resolved. In the future, please use one of our issue templates. This greatly helps us to address your issue more quickly.

Thanks and glad your issue is resolved,
Adam