WEC-Sim/WEC-Sim #859: [Theory or Implementation] Crane ship simulation Capytaine -> Wec-SIM

Dadidal, 2022-04-28:
Hello! Could you please explain me what kind of simulation in BEAM software I should perform to make a correct model in Wec-SIM. The problem is described here:

I want to make a simulation of crane ship using Capytaine (BEM software) -> Wec-SIM. My crane ship consists of 4 parts presented in figure below (highlighted by different colors). Part "Hull" is floating part (hydrodynamic body) and other parts ("Bridge", "Crane" and "Boom") are Nonhydrodynamic bodies. The draft of Hull ONLY is 0.04 m, but the draft of FULL model is 0.055 m. Center of Gravity of Hull is different from Center of Gravity of Full model. In Capytaine I analyze only HULL part and want to add other parts in Wec-SIM as Nonhydrodynamic bodies with specified mass and Inertia (I attached the SIMULINK model). 
I don't understand the following:

1. What draft should I analyze in Capytaine - 0.04 or 0.055 m? As I understand since the future model in Wec-SIM will be 0.055m I have to calculate the case 0.055m, right?
2. What Center of Gravity I should set in Capytaine settings and therefore to Hydrostatics.dat file for BEMIO? - CoG of HULL ONLY or CoG of FULL ship? This is difficult to understand for me because Nonhydrodynamic bodies in Wec-SIM will change my CoG but the Hydroforces and Hydrostatic stiffness calculated in Capytaine have to correspond the Full model before the use of BEMIO.
3. How I have to specify parameters of hydrodinamic body "Hull" wecSimInputFile.m?
- Since draft and displaced volume of model analyzed in Capytaine correspond to FULL model (0.055 m) but Nonhydrodynamic bodies also have masses what mass I have to set in Wec-SIM for "Hull" part? Am I right that option  _body(1).mass = 'equilibrium';_ is not correct in this case and I have to set mass manually equal to weight of Hull part only?
- What Inertia [Ixx Iyy Izz] I have to set for Hull part?

[image]
[image]

akeeste, 2022-05-03:
Hi @Dadidal 

Thanks for opening an issue. I see that you have all of the required information, but please maintain the template formatting. This helps our team find information and solve issues quickly.

There are multiple ways to set-up your model. I suggest that you use the least number of bodies possible. For example, if the Bridge and Hull are fixed together and do not move relative to one another, then you could consider them as one body. Most importantly though, I highly recommend that you start with short Capytaine runs and test [WEC-Sim noWave cases](https://wec-sim.github.io/WEC-Sim/master/user/troubleshooting.html#numerical-test-cases) to ensure that your initial positions and hydrostatic stiffness are accurate. 

I will answer your below questions assuming that there are four bodies as you show above:

> 1. What draft should I analyze in Capytaine - 0.04 or 0.055 m? As I understand since the future model in Wec-SIM will be 0.055m I have to calculate the case 0.055m, right?

In the BEM code, you are correct that only the hull needs to be represented because the other bodies are above the still water line. The hull should be given the draft of the full case, because this is where the real equilibrium point will be when all bodies are included. So use a draft of 0.055m. 

> 2. What Center of Gravity I should set in Capytaine settings and therefore to Hydrostatics.dat file for BEMIO? - CoG of HULL ONLY or CoG of FULL ship? 

Likewise the Hull cg should only correspond to the hull geometry, with a draft of 0.055m. The hydroforces can be applied at the hull cg because only the hull is important for the BEM calculation. You are correct that the nonhydrodynamic bodies affect the overall cg, but this change is accounted for in WEC-Sim when defining the nonhydrodynamic bodys' mass and inertia.

> 3. How I have to specify parameters of hydrodynamic body "Hull" wecSimInputFile.m?
> 
> * Since draft and displaced volume of model analyzed in Capytaine correspond to FULL model (0.055 m) but Nonhydrodynamic bodies also have masses what mass I have to set in Wec-SIM for "Hull" part? Am I right that option  _body(1).mass = 'equilibrium';_ is not correct in this case and I have to set mass manually equal to weight of Hull part only?

Correct, if your nonhydro bodies have some mass and no displaced volume, then you cannot use `body(1).mass = 'equilibrium'`. You have to set the mass manually for the Hull part only. 

> * What Inertia [Ixx Iyy Izz] I have to set for Hull part?

You will have to calculate or approximate the inertia given some mass distribution within the hull. For example to start, you could assume that the hull has a uniform density, or mass is concentrated in the outer shell. Meshmagick [plain mesh inertia](https://lheea.github.io/meshmagick/api/meshmagick.mesh.html#meshmagick.mesh.Mesh.eval_plain_mesh_inertias) and [shell mesh inertia](https://lheea.github.io/meshmagick/api/meshmagick.mesh.html#meshmagick.mesh.Mesh.eval_shell_mesh_inertias) functions can help you do this.

I hope this helps. Please let me know if you have any more questions.
Adam

akeeste, 2022-05-13:
Hi @Dadidal

Did the above response help clarify your WEC-Sim implementation? Please let me know if you have any more questions. Unfortunately we can't provide detailed support on BEM codes, but can help you set-up your WEC-Sim model accordingly.

Adam

akeeste, 2022-06-03:
Hi @Dadidal 

I hope this helps your problem. I will close this issue now. If you have additional problems, please submit a new issue.

Thanks,
Adam

closed 2022-06-03 by akeeste