WEC-Sim/WEC-Sim #860: [BUG] WEC-Sim Applications : PTO-Sim

jacksder, 2022-05-04:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
Sorry if this is the wrong repo to post this issue, but I didn't see an issues option for the PTO-Sim_Applications repo. The "ABC2DQ" and "DQ2ABC" transformation blocks used in the RM3_DD_PTO directLinearGenerator are incorrect.

**WEC-Sim file(s)**
file "WEC-Sim_Applications/PTO-Sim/RM3/RM3_DD_PTO/RM3_DD_PTO.slx" (https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/PTO-Sim/RM3/RM3_DD_PTO)

**To Reproduce**
_Steps to reproduce the behavior:_
1. Open the "RM3_DD_PTO.slx" file.
2. Navigate within the "PTO-Sim" subsystem.
3. Look under mask of "Direct Drive Linear Generator".
4. View the "ABC2DQ" and "DQ2ABC" subsystems. Specifically the order of inputs of the first MUX block.

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
N/A

**Expected behavior**
The correct abc-dq and dq-abc transformations are to be expected.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release] 2021b
 - [WEC-Sim Release] v4.4

**Additional context**
N/A

dforbush2, 2022-05-05:
Hi @jacksder 

Good catch. It looks like the mux inputs are flipped from what they ought to be. I will check this out and implement a fix today, 

Just FYI: there is  a major overhaul to PTO sim coming with the V5.0 release.

jacksder, 2022-05-05:
Thanks for the heads up on the major update to PTO sim. Is there any reading material you can point me to on what will be changed? Or will that be available once V5.0 is released?

dforbush2, 2022-05-05:
Be aware that it is a work-in-progress until it is merged with master, but some documentation is on the dev branch. http://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#pto-pto-sim

dforbush2, 2022-05-05:
Hi @jacksder 

I'm having what I think are server issues so I can't push right now...but I wanted to get you rolling. I've attached a version of the PTO-Sim library with what I hope is the fix implemented. The input order to the ABC2DQ and DQ2ABC blocks were off (2 each) in the rotary and linear generator blocks. 

Thank you for finding this
[PTOSim_Lib.zip](https://github.com/WEC-Sim/WEC-Sim/files/8636476/PTOSim_Lib.zip)
!

jacksder, 2022-05-06:
Great, thanks for sharing the implemented fix!

kmruehl, 2022-05-12:
@dforbush2  and @jleonqu thank you!

commit fecc5cdb (2022-05-12): ptosim bugfix resolves #860
closed 2022-05-12 by kmruehl