WEC-Sim/WEC-Sim #861: Matlab command window shows：The index exceeds the number of array elements.

woosalute, 2022-05-04:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
I'm sorry my English level may not be too good and that I'm a wec-Sim newbie. I imitated the example RM3, only built a float of different sizes, and then used aqwa to solve the float's hydrodynamic files lis and ah1, but when using wecsim simulation, matlab reported an error:, 
                                                                  Incorrect use wecSim (line 44)
                                                                 The index exceeds the number of array elements (1).
and the mechanics explorer can not output the image normally, I don't know how to solve this problem now, please answer it, I will be grateful.

**WEC-Sim file(s)**
I put all the simulation files in a compressed package and uploaded them to you.

**To Reproduce**
_Steps to reproduce the behavior:_
1.I built a new simulink model(maybe it is incorrect)
2.I removed the definition statement about the sheet and removed the Flating (3DOF) Joint in the wecSiminputFile.m
3. input 'wecSim' in command window
4. then the error appears(as it is shown in the picture):
                                                                     Incorrect use wecSim (Line 44)
                                                                    Error evaluating 'StopFcn' callback of Frames block (mask) 'RM4/Global Reference Frame'.
                                                              Callback string is 'run('stopWecSim')'
                                                   Caused by:
                                                               Incorrect use  wecSim (第 44 行)
                                                                The index exceeds the number of array elements (1).

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
Finally, it should output six response images and an animation of a float moving on the surface of the water in Mechanics Explorer.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: **Windows 10**
 - [MATLAB/Simulink Release]  **2021a**
 - [WEC-Sim Release] **v4.4**

**Additional context**
because i am a newbie,so please tell me the details about how to deal with these problems,it is a part of my Master's thesis so i am eager to hear from you! thank you very much!
[RM4.zip](https://github.com/WEC-Sim/WEC-Sim/files/8619901/RM4.zip)

dforbush2, 2022-05-04:
Hi @woosalute 

Based on that error message, your model is running, but the stop function is failing. This is the 'userDefinedFunctions.m' included in your zip file. This file still references the second WEC body: 

```
%Plot heave response for body 1
output.plotResponse(1,3);

%Plot heave response for body 2
output.plotResponse(2,3);

%Plot heave forces for body 1
output.plotForces(1,3);

%Plot heave forces for body 2
output.plotForces(2,3);
```

So because you are only using one body in your simulation, this code throws your error when it tries to run. Remove the portions of this file that reference the 2nd WEC body and everything runs fine!

Some additional feedback that might help get you started:
1) Unless you are planning to just run regular wave tests at exactly your BEM frequencies, it is a good idea to calculate your hydrodynamic coefficients in BEM using ~100 frequencies, at a minimum. This is because the impulse response function is used for 'regularCIC' or 'irregular' wave types, and this impulse response function likely is not well-approximated with so few frequencies.
2). 'Equilibrium' mass flags are a good tool to use for debugging, but if you are doing a geometry design, it is good to ensure that your body.dispVol and body.mass are as expected. 
3). When debugging, it is sometimes useful to run your model using the Simulink GUI. Place a debug point on line 46 of wecSim.m. When the code stops here, you can open your *slx model and run it using the buttons in the Simulink GUI, which I find is useful to catch bugs like this.
4). If you do not have a 'userDefinedFunctions' in your model directory, then, depending on your MATLAB path configuration, wecSim may try and execute a different file called 'userDefinedFunctions'  in another directory. This will frequently cause errors, so during debugging, I frequently place an empty *.m file called userDefinedFunctions in my model directory.

woosalute, 2022-05-05:
> Hi @woosalute
> 
> 你好@woosalute
> 
> Based on that error message, your model is running, but the stop function is failing. This is the 'userDefinedFunctions.m' included in your zip file. This file still references the second WEC body:
> 
> 根据该错误消息，您的模型正在运行，但是 stop 函数失败。这就是 zip 文件中包含的 userDefinedFunctions.m。此文件仍然引用第二个 WEC 主体:
> 
> ```
> %Plot heave response for body 1
> output.plotResponse(1,3);
> 
> %Plot heave response for body 2
> output.plotResponse(2,3);
> 
> %Plot heave forces for body 1
> output.plotForces(1,3);
> 
> %Plot heave forces for body 2
> output.plotForces(2,3);
> ```
> 
> So because you are only using one body in your simulation, this code throws your error when it tries to run. Remove the portions of this file that reference the 2nd WEC body and everything runs fine!
> 
> 因此，由于在模拟中只使用一个主体，所以当该代码试图运行时会抛出错误。删除该文件中引用第二个 WEC 主体的部分，一切运行正常！
> 
> Some additional feedback that might help get you started:
> 
> 一些额外的反馈可能会帮助你开始:
> 
> 1. Unless you are planning to just run regular wave tests at exactly your BEM frequencies, it is a good idea to calculate your hydrodynamic coefficients in BEM using ~100 frequencies, at a minimum. This is because the impulse response function is used for 'regularCIC' or 'irregular' wave types, and this impulse response function likely is not well-approximated with so few frequencies. 除非你计划在边界元法频率上进行常规波浪测试，否则最好在边界元法中至少使用100个频率来计算你的水动力系数。这是因为脉冲响应函数用于“正则化 cic”或“不规则”波型，这种脉冲响应函数可能不能很好地近似于这么少的频率
>    2). 'Equilibrium' mass flags are a good tool to use for debugging, but if you are doing a geometry design, it is good to ensure that your body.dispVol and body.mass are as expected. 2).“均衡”质量标志是一个很好的调试工具，但如果你正在进行几何设计，最好确保你的身体。 dispvol 和 body.mass 符合预期
>    3). When debugging, it is sometimes useful to run your model using the Simulink GUI. Place a debug point on line 46 of wecSim.m. When the code stops here, you can open your *slx model and run it using the buttons in the Simulink GUI, which I find is useful to catch bugs like this. 3).在调试时，有时使用 Simulink GUI 运行模型是有用的。在 wecSim.m 的第46行上放置一个调试点。当代码在这里停止时，您可以打开 * slx 模型并使用 Simulink GUI 中的按钮运行它，我发现这对捕获这样的 bug 很有用
>    4). If you do not have a 'userDefinedFunctions' in your model directory, then, depending on your MATLAB path configuration, wecSim may try and execute a different file called 'userDefinedFunctions'  in another directory. This will frequently cause errors, so during debugging, I frequently place an empty *.m file called userDefinedFunctions in my model directory. 4).如果你的模型目录中没有一个 userDefinedFunctions，那么，根据你的 MATLAB 路径配置，wecSim 可能会尝试并在另一个目录中执行一个名为 userDefinedFunctions 的不同文件。这会经常导致错误，所以在调试过程中，我经常放置一个空 * 。在我的模型目录中，一个名为 userDefinedFunctions 的 m 文件

Thank you for your quick answer! Next, I will carry out the simulation again according to the method you said, and feed back the results to you in time. At the same time, thank you for your sincere suggestions. I will continue to study WEC SIM and communicate with you

woosalute, 2022-05-06:
At your suggestion, I deleted the statement about body(2) in the userDefinedFunction .m and completed the simulation, but at the same time a new problem appeared: in the pop-up Mechanics Explorers I can only see the water surface and the seabed, and I can't see my model, but when I click the "Fit to view" button, my model will appear in the window, but I can't see the water surface and the seabed at this time. I don't know exactly what caused this or if you're having the same problem when you run it, can you help me answer that? Thank you very much!
[RM4.zip](https://github.com/WEC-Sim/WEC-Sim/files/8638174/RM4.zip)
[image]
[image]

woosalute, 2022-05-10:
@dforbush2 
Dear sir，The problem I encountered in the model display is still not solved, can you give me some suggestions?Thank you very much！

dforbush2, 2022-05-10:
Hi @woosalute 

I took another look. This seems to be an issue with your STL model. I'm guessing that you modeled this is Solidworks? This is a common issue where Solidworks does not always save the units modeled as intended unless very robustly specified. I'm guessing that millimeters were the intended unit, but it was saved as meters. As such, it is 40 KM in diameter!

Second, the geometry is not modeled correctly about (0,0,0). This can be difficult with solidworks, but recall the WEC-Sim sign convention that origin is at the center of the device in the still-water free surface. The center of your device seems to be closer to (19500, 19500), this is why it does not appear in the default Explorer view. 

To fix this, I would recommend either:
1) Re-making the Solidworks model with the proper origin and model using Meters.
2) Use a tool like MeshMagick to transform your STL file directly. https://lheea.github.io/meshmagick/install/install.html

But be aware: as long as your geometry is correct in your H5 file, the hydrodynamics are correct. Unless you are using non-linear hydrodynamics (simu.nlHydro = 2), the STL file is used only for visualization and will not affect your hydrodynamics. Looking at your H5 file I see a reasonable CG and displacement volume, so it seems to me like this is correct but it is worth double-checking the geometry you have used in AQWA.

closed 2022-05-17 by dforbush2