WEC-Sim/WEC-Sim #863: Model Problems:The image can be output, but the model created by myself cannot be displayed normally.

woosalute, 2022-05-08:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
     This is the second time I have asked my question in this column, if the weC-Sim developers can solve my problem I would be grateful, I built my own model through the modeling software, and was able to simulate successfully.But there is a problem: in the pop-up mechanics explorers, I can only see the water surface and seabed, but I can't see my model. However, when I click the "fit to view" button, my model will appear in the window, but I can't see the water surface and seabed at this time. In addition, in the whole simulation time, my model did not move significantly, I don't know what caused this phenomenon.
 can you help me answer this question? thank you very much!

**WEC-Sim file(s)**
I put all the simulation files in a compressed package and uploaded them to you.

**To Reproduce**
_Steps to reproduce the behavior:_
1.I built a new simulink model(maybe it is incorrect)
2.I removed the definition statement about the sheet and removed the Flating (3DOF) Joint in the wecSiminputFile.m
3. input 'wecSim' in command window
4. in the pop-up Mechanics Explorers I can only see the water surface and the seabed, and I can't see my model
[image]
    But when I click the "Fit to view" button, my model will appear in the window, but I can't see the water surface and the seabed at this time.
[image]

**Expected behavior**
I can directly see my model oscillating on the water

**Screenshots**
I've pasted them on the front to see。

**Workstation Information (this section MUST BE completed):**
OS: Windows 10
[MATLAB/Simulink Release] 2021a
[WEC-Sim Release] v4.4
**Additional context**
Through the solution of the last problem, I have gradually liked learning WEC SIM and am in high interest, so please tell me the details of how to deal with these problems. This is a part of my master's thesis, so I am eager to hear from you! thank you very much! In addition, I would appreciate it if you could give me some suggestions on modeling！
[RM4.zip](https://github.com/WEC-Sim/WEC-Sim/files/8647226/RM4.zip)

reos-rcrozier, 2022-05-08:
I'm not a developer, just a user. Just a guess, without looking at anything: the scale of the STL file is wrong, or the location of the centre of the object STL file is wrong (i.e. not a the centre of mass of your wecSim body).

woosalute, 2022-05-09:
Thank you very much for your answer, do you mean that there is a problem with my stl file? I used AQWA to get this file but I don't quite understand how to modify it, how do I need to set it up?  Thank you very much!

> I'm not a developer, just a user. Just a guess, without looking at anything: the scale of the STL file is wrong, or the location of the centre of the object in STL file is wrong (i.e. not a the centre of mass of your wecSim body).

woosalute, 2022-05-09:
@reos-rcrozier 
Thank you very much for your answer, do you mean that there is a problem with my stl file? I used AQWA to get this file but I don't quite understand how to modify it, how do I need to set it up?can you help me see it? Thank you very much!

reos-rcrozier, 2022-05-09:
Yes this is known as the geometry file in wecSim, decribed in the manual [here](https://wec-sim.github.io/WEC-Sim/master/user/workflow.html#geometry-file). I don't know if the dimensions of the STL are supposed to be in mm or m, but if you have the wrong one, the visualisation will look wrong. If the STL isn't created such that the point (0,0) in the STL file is the centre of gravity of the body it will look wrong in the visualization.

A bit of googling will reveal many ways to manipulate STL files.

I don't know for sure this is the problem, just a suggestion.

woosalute, 2022-05-10:
@kmruehl 
Dear developer，i am Sorry to bother you.The problem I encountered in the model display is still not solved, can you give me some suggestions?Thank you very much！

reos-rcrozier, 2022-05-11:
told you so

woosalute, 2022-05-14:
@dforbush2 
Dear Sir，Like you said, the geometric model was built in  solidworks, and I will re-tweak the model as you suggest and check the visualization problems of the model, thank you very much for your help!

woosalute, 2022-05-14:
@reos-rcrozier 
Thank you for your generous advice

closed 2022-05-17 by dforbush2