WEC-Sim/WEC-Sim #867: [Developer Issue]

jtgrasb, 2022-05-17:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
N/A

**Describe the issue**
This issue is related to issue #861 (https://github.com/WEC-Sim/WEC-Sim/issues/861#issuecomment-1117708162). Currently, the post-processing (run('stopWecSim')) is called by the stopFcn callback of the global frames block. When there are errors in any of the files associated with post-processing, the specific line of error is not shown because the files are called by Simulink, making it hard to debug any errors. 

A potential solution of this is to move the run('stopWecSim') command outside of Simulink and just put it at the end of the wecSim.m file. This seems like a simple solution, so there is likely a reason this hasn't been done yet. Are there any issues with this solution? - the tests all pass when I implement this.

jtgrasb, 2022-05-18:
I looked into more potential fixes for this issue. 

First, the try-catch ([https://www.mathworks.com/help/simulink/ug/error-handling-in-simulink-using-mslexception.html](url)) method doesn't seem to help, because Simulink receives the same ambiguous errors when trying the stopWecSim.m script, providing no way to get any more error messages.

Next, the diagnostic messages ([https://www.mathworks.com/help/simulink/ug/customize-simulation-error-messages.html](url)) could be helpful to make specific checks on the post-processing, but since we can't anticipate every error that the user would make, this wouldn't be very encompassing.

I also tried making the stopWecSim function a MATLAB Function block that checks for when the time is equal to endTime, but Simulink seems to not support many of the MATLAB functions such as clear because they aren't supported for code generation.

It seems to me the best option would be to move the stopWecSim command to the wecSim.m script, and rework the run from Simulink process to include running stopWecSim after simulation.

dforbush2, 2022-06-06:
See #873 for an odd error to be aware of.

closed 2022-07-06 by kmruehl