WEC-Sim/WEC-Sim #872: Error message in wecsiminput.m file in WEC-Sim 4.4 version

Aiswariak, 2022-05-24:
I have been using WEC-Sim 4.2 version and has now updated to 4.4 version. My MATLAB is R2021b. When I am trying to run wecsiminputfile.m, I get an error message as follows:

Unrecognized function or variable 'simFile'.

Error in wecSim (line 46)
sim(simFile, [], simset('SrcWorkspace','parent'));

The same code works fine with the older version. I could also run RM3 tutorial successfully. Kindly help @kmruehl.
[image]

_Originally posted by @Aiswariak in https://github.com/WEC-Sim/WEC-Sim/issues/255#issuecomment-1134242989_

dforbush2, 2022-05-24:
Hi @Aiswariak 

Would you please re-post your issue using one of the issue templates? This helps us better address your issue quickly. I would also recommend you upload your run directory as a *zip.

Best,
Dom

dforbush2, 2022-05-24:
In Version 4.4: line 45 of wecSim.m should read:

`sim(simu.simMechanicsFile, [], simset('SrcWorkspace','parent'))'

It appears that your path may still be pointing to your old WEC-Sim installation? Pulling the latest version of the code, and ensuring your matlab path is updated appropriately, will solve your issue.

closed 2022-05-26 by dforbush2

khushal143, 2023-12-30:
when i type in the command prompt of Matlab
[image]
 wecSim
i faced the attached error messages

khushal143, 2023-12-30:
I am facing the attached error after typing wecSim in command window How to fix this problem the path has already added 
[image]

dforbush2, 2024-01-02:
Hello @khushal143 

Looking at your directory structure, I suspect you have attempted to install a static code download, rather than a git pull? If this is the case, please open another issue, since the solution is likely to be unique. If not, and you have properly installed wecSim and added your path, this error suggests that either: 
1) your current working directory is not correct. Based on your directory structure, you should be in C:\Users\HP Downloads\Compressed\WEC-Sim-master when you run the `wecSim` command
2) that rm3.slx does not exist in C:\Users\HP Downloads\Compressed\WEC-Sim-master

Please refer to Step 5 http://wec-sim.github.io/WEC-Sim/master/user/workflow.html#running-wec-sim. Your $CASE directory should contain your wecSimInputFile.m and your rm3.slx for the case you wish to run. As an example of $CASE directory, see WEC-Sim/examples/RM3