WEC-Sim/WEC-Sim #873: [Theory or Implementation]

Aiswariak, 2022-05-25:
I have been using WEC-Sim 4.2 version and has now updated to 4.4 version. My MATLAB is R2021b. When I am trying to run wecsiminputfile.m, I get an error message as follows:

Unrecognized function or variable 'simFile'.

Error in wecSim (line 46)
sim(simFile, [], simset('SrcWorkspace','parent'));

I am reposting this as per your suggestion @ https://github.com/dforbush2
https://github.com/WEC-Sim/WEC-Sim/issues/872#issue-1246015653

The same code works fine with the older version. I could also run RM3 tutorial successfully. Kindly help
[image]

Aiswariak, 2022-05-25:
Hi @ [dforbush2](https://github.com/dforbush2),
I believe that I have  properly updated the matlab path and can run RM3 tutorial successfully too.

The line 45 of wecSim.m is:

`sim(simuFile, [], simset('SrcWorkspace','parent'))'
Does this mean that my path is still pointing to old WEC-Sim installation?

I tried changing this line to :
`sim(simu.simMechanicsFile, [], simset('SrcWorkspace','parent'))'. 
Now its running and I am getting the output folder. Still it shows some error at the end as follows:

Error using wecSim (line 46)
Error evaluating 'StopFcn' callback of Frames block (mask) 'RHP1610/Global Reference Frame'.
Callback string is 'run('stopWecSim')'
Caused by:
    Error using wecSim (line 46)
    Unrecognized function or variable 'projectRootDir'.

Does this effect my results?

I have attched the zip folder for your reference.

[T=1.6s, H=0.10m.zip](https://github.com/WEC-Sim/WEC-Sim/files/8768383/T.1.6s.H.0.10m.zip)

dforbush2, 2022-05-26:
I'm browsing the repository at the v4.4 release. In wecSim.m I see:

```
% Initialize WEC-Sim
run('wecSimInputFile');
simFile = simu.simMechanicsFile;
clear simu waves body cable pto constraint ptosim mooring 
clear simu waves body pto constraint ptosim mooring 

runWecSimCML = 1;
run('initializeWecSim');
sim(simFile, [], simset('SrcWorkspace','parent'));

```

Note that this has sim(simFile), not sim(simuFile), and simFile = simu.simMechanicsFile; as per the earlier line. Unless this was a typo, this means you had changed the source file somehow, and this has caused your initial error. More recent WEC-Sim editions use simu.simMechanicsFile as the argument to "sim" directly because modern WEC-Sim versions allow fields of a structure to be passed as arguments directly (where as older versions did not. My earlier comment was mistaken: this change was not present in v4.4 upon release). 

When you change the wecSim.m file and are able to run until the userDefinedFunctions tries to execute, your run is completed: this error does not affect the simulation results. The error results from stopWecSim looking for a variable named 'projectRootDir' but being unable to find it. 

This variable should be put into your workspace by the simulink model when it finishes running: this suggests that though you have set up your path to look at the new source code, your simulink library may be the old version. I recommend you 1). delete any *Lib.slx files from previous versions and 2). repeat step 3 of http://wec-sim.github.io/WEC-Sim/master/user/getting_started.html to ensure your simulink library is referencing v4.4 blocks.

Hope that helps!

dforbush2, 2022-05-26:
I also do recommend that you consider using the Git Pull installation procedure: this way you will get all of the bug fixes and updates within a single release rather than the download of the static code.

Aiswariak, 2022-05-31:
I have installed wecSim using GitPull installation procedure and still I am getting error as follows:

Error using wecSim (line 45)
Error evaluating 'StopFcn' callback of Frames block (mask) 'RHP1410/Global Reference Frame'.
Callback string is 'run('stopWecSim')'
Caused by:
    Error using wecSim (line 45)
    Index exceeds the number of array elements. Index must not exceed 1.

 I am unable to fix it. Kindly help

I am attaching the zip file and a screenshot for your reference
[image]
[varying chamber.zip](https://github.com/WEC-Sim/WEC-Sim/files/8803045/varying.chamber.zip)
.

dforbush2, 2022-06-01:
Hi there,

From your screenshot, this is because you do not have a userDefinedFunctions.m in your path. Because of this, WEC-Sim is looking elsewhere, and it is finding one (likely example/RM3) that is looking for the outputs of 2 bodies. It is failing, because you only have 1 body.

I recommend putting back in your old userDefinedFunctions.m (included in an earlier *zip), or simply a blank file called userDefinedFunctions.m in this directory.

Aiswariak, 2022-06-06:
Hi @dforbush2 ,

I have added userDefinedFunctions.m in my path. But I am getting another error as follows:

" Error using wecSim (line 45)
Error evaluating 'StopFcn' callback of Frames block (mask) 'RHP1410/Global Reference Frame'.
Callback string is 'run('stopWecSim')'
Caused by:
    Error using wecSim (line 45)
    Unrecognized function or variable 'projectRootDir'.

In the mechanical explorer, my body is static even though I am getting force and response graphs.

In the simulink model, I could see these errors:

Variant control 'sv_visualizationOFF' of variant block '[RHP1410/Global Reference Frame/waveVis](matlab:open_and_hilite_hyperlink ('RHP1410/Global Reference Frame/waveVis','error'))' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model. Ensure that the condition does not match the model name.

Variant control 'sv_visualizationON' of variant block '[RHP1410/Global Reference Frame/waveVis](matlab:open_and_hilite_hyperlink ('RHP1410/Global Reference Frame/waveVis','error'))' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model. Ensure that the condition does not match the model name.

I could find find similar issue in #144 and tried to fix it as per that, but failed. Kindly help. I really need to work using WecSim.

[varying chamber.zip](https://github.com/WEC-Sim/WEC-Sim/files/8842347/varying.chamber.zip)
[image]

Thank you.

Aiswariak, 2022-06-07:
Hi @dforbush2 ,

Thank you for the quick response and clear explanation. I would still have some queries regarding the above points:

1. I believe that I have properly set up the path  and have run addWecSimSource.m as per the reference provided. I have checked that again. Still I get this message:

Unrecognized function or variable 'wecSimInitialize'.

How to make sure that the appropriate version of the Simulink library is being referenced? Can you guide me on this or provide any references as I am new to this?

3. I removed the 'clear' from the code and this error is removed.

4. I have run the same case in my old wecSim version 4.2 in another system and am able to get the movement of body in mechanical explorer and the graphs properly. The graphs obtained in that case is more believable as it has the same pattern as that of the input wave elevation whereas here I am getting some odd graphs. The values are different in both versions.

I really want and hope to work in the new version. Kindly help.

Thank you

dforbush2, 2022-06-07:
So I am a bit unclear...if you can run 3) successfully, it seems like you are up and running? If this is the case your path and simulink library are working properly. 

We can look into the disagreement between v4.2 and v5.0: as you suggest the movement here is at lower frequencies than one might expect. 

Some things to try:
1). switch to a variable step solver, like ode45. 
_same results pretty much as before_
2). instead of a user-spectrum import, try a regular wave excitation with 'regularWaveCIC' at your test frequencies.
I tried this, and results look good!
3) Do the same with irregular spectra. I tried this too, and things look alright. 

This suggests that either there is something strange in your BEM results at low frequencies, or there is a bug in the v5.0 implementation of the waveClass elevationImport function. You BEM results actually look really nice, aside from some spikiness at the lowest frequencies, so leads me to believe there might be a bug here. Will follow up.

Aiswariak, 2022-06-08:
Hi @dforbush2 ,

Thank you so much for the timely response. I too tried 1,2 and 3 mentioned above and it is properly working as you said.  So, even I think the problem is with waveClass elevationImport function. With the same BEM results at low frequencies, it is working for regular and irregular wave classes. And even it worked fine with version 4.2 for the imported elevation. 

Kindly follow up. Thank you.

akeeste, 2022-06-14:
Hi @Aiswariak 

I have just been pulled into this issue and want to clarify where you are at. To my understanding:
- Your case is now running without errors or bugs
- There seems to be a difference between v4.2 and v4.4 ``elevationImport``
- You are now using the varying chamber case

Can you please confirm that this is all true or give corrections?

Thanks,
Adam

Aiswariak, 2022-06-15:
Hi @akeeste 

1. My case is running without errors and bugs
2. There is difference between latest version and v4.2 elevation import and results also. I am getting different results in both versions. Also, results from v4.2 seems to be reliable as it follows the pattern of wave elevation imported.
3. I am now using a varying chamber case now. It is just that the imported elevations are obtained inside a chamber experimentally by varying the chamber width. The zip file attached is for one such case of chamber width.

Thank you.

Aiswariak, 2022-06-16:
Hi @akeeste ,

I have attached the results of my simulations of this particular issue in both versions.
[Elevation import graphs.pdf](https://github.com/WEC-Sim/WEC-Sim/files/8917893/Elevation.import.graphs.pdf)

akeeste, 2022-06-17:
Hi @Aiswariak,

Thank you for attaching those plots. I will try to recreate the issue and investigate what is happening.

Adam

kmruehl, 2022-06-22:
@Aiswariak I've been working with @dforbush2 and @salhus to address this issue. We've been able to reproduce the issue running the [varying chamber case you provided](https://github.com/WEC-Sim/WEC-Sim/files/8842347/varying.chamber.zip) with the [latest version of WEC-Sim on the master branch](https://github.com/WEC-Sim/WEC-Sim). 

However, we were unable to reproduce the issue using the [RM3 example](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/RM3) with the [latest version of WEC-Sim on the master branch](https://github.com/WEC-Sim/WEC-Sim) (a WEC-Sim case we know to be stable).  This led us to think that the issue may be with the combination of wave data and the device model for the varying chamber case. I've attached an updated model of the same device ([varying chamber_RM3data.zip](https://github.com/WEC-Sim/WEC-Sim/files/8961597/varying.chamber_RM3data.zip)) using the elevation data from the RM3 example. I reduced the wave height from the RM3 example to be comparable to the wave height of the data you provided, and the case is stable. 

We haven't ruled out that there may be a bug in the convolution integral calculation for `elevationImport`, but before digging into it further we'd like to better understand the case you provided. Where did that wave data come from? And how was the PTO damping value chosen?

Aiswariak, 2022-06-23:
Hi @kmruehl,
Even I am using the latest version of WEC-Sim on the master branch now. The updated model which you have sent (varying.chamber_RM3data.zip) worked well for me also. But I need to run the simulation corresponding to my wave elevation data. That data is obtained from the experiments which we have conducted in our lab which is the wave elevation that passes through a porous wall. I have done simulations for various values of PTO damping from 10 to 50 Ns/m and chosen the optimum value corresponding to maximum power.

dforbush2, 2022-06-23:
Hi @Aiswariak 

Thank you for the explanation. I understand the desire to use your device and sea state. However, if you set damping to zero, you will see that your device is unstable in simulation. A damping coefficient of 20 is likely only viable because of the instability. Because of this instability, we do not expect typical device behavior (i.e., frequency of motion to match dominant frequency of excitation), so it is difficult to debug this issue in the context of your model. 

To fix this instability please double-check the hydrostatics of your model. First, I would double check that your *.gdf mesh is closed. Taking a screenshot from Rhino, I see that only 1/4 of the device is modeled: 
[image]

Opening the *gdf as a text file, however, I see that the symmetry flags are zero.
0  0 	ISX  ISY

I would expect both of these entries to be 1 (not zero) based upon the Rhino screenshot. It is possible that you have fixed this in your NEMOH *dat file (I am not as familiar with NEMOH)...but please do double check. Please also ensure that your CG location is correctly specified relative to the free-surface for your BEM run. Looking at your *H5 file and comparing with the cross-section of your mesh intersecting the free-surface, I would expect your hydrostatic restoring coefficient in heave to be about twice what it is. 

This might not fix the issue but should improve your model!

dforbush2, 2022-06-30:
Hi all,

Following another extensive debug session, what I have come up with is that the 'etaImport' option is extremely sensitive to time-step selection and each wave elevation data point in sea-states that are nearly mono-chromatic and at a frequency that is not sufficiently low (your waveP1 is nearly monochromatic). This was validated for the RM3 example case which we know to be stable. 

Frankly, convolution integral functions, approximated in the way that they are in BEM solvers or any other numerical approximation, are poor choices as models for nearly monochromatic sea states, since the error in estimated excitation at the predominant frequency is likely to be high (compared to a broad-band, irregular excitation sea-state, where the aggregate of the errors for each excitation component is more likely to be close to zero): there is nothing "wrong" with the implementation (excepting the minor bugs @akeeste, @salhus @kmruehl and myself discovered when working through this). I am still getting to the bottom of why this worked better in v4.2, but I suspect it is not innately more stable.   

An alternative recommended workflow is to switch out the etaImport for an equivalent spectrumImport w/ seeded phase.

Aiswariak, 2022-07-01:
> Hi @Aiswariak
> 
> Thank you for the explanation. I understand the desire to use your device and sea state. However, if you set damping to zero, you will see that your device is unstable in simulation. A damping coefficient of 20 is likely only viable because of the instability. Because of this instability, we do not expect typical device behavior (i.e., frequency of motion to match dominant frequency of excitation), so it is difficult to debug this issue in the context of your model.
> 
> To fix this instability please double-check the hydrostatics of your model. First, I would double check that your *.gdf mesh is closed. Taking a screenshot from Rhino, I see that only 1/2 of the device is modeled: [image]
> 
> Opening the *gdf as a text file, however, I see that the symmetry flags are zero. 0 0 ISX ISY
> 
> I would expect the first of these entries to be 1 (not zero) based upon the Rhino screenshot. It is possible that you have fixed this in your NEMOH *dat file (I am not as familiar with NEMOH)...but please do double check. Please also ensure that your CG location is correctly specified relative to the free-surface for your BEM run. Looking at your *H5 file and comparing with the cross-section of your mesh intersecting the free-surface, I would expect your hydrostatic restoring coefficient in heave to be about twice what it is.
> 
> This might not fix the issue but should improve your model!

Hi @dforbush2 ,
I have modelled only the half geometry as it is symmetrical and Nemoh requires only half of it. But I am not aware of the symmetry flags when opened as a text file. I too obtained it as 0 0 ISX ISY. I will double check this.

Thank you

Aiswariak, 2022-07-01:
Hi @dforbush2 ,
As the etaImport values are obtained from experiment conducted here, I am not sure whether I can use an equivalent spectrumImport w/ seeded phase. 
So, do you mean to say that the results obtained from v4.2 is not believable? Would that also contain errors?

dforbush2, 2022-07-01:
Hi @Aiswariak 

To my knowledge, NEMOH *dat mesh has a flag for XOz symmetry, and the second element of line 1 in your *dat file should be a 1. All I can say for sure is that your hydrostatic stiffness in heave is half of what it is by a hand calculation. I reviewed the NEMOH documentation https://lheea.ec-nantes.fr/valorisation/logiciels-et-brevets/nemoh-mesh

I am digging into the version difference this morning. I wouldn't say that the results are necessarily "incorrect", but that your modeling approach is not robust for nearly monochromatic waves.

dforbush2, 2022-07-01:
I am unable to observe a significant difference between v5.0 and v4.2, so I stand behind my earlier remarks. I think this exemplifies the limitations of this solver method. I will produce an example for you of how to do an equivalent study w/ spectrumImport.

Aiswariak, 2022-07-04:
Hi @dforbush2,

I have also gone through the NEMOH documentation and corrected it. Thank you so much for identifying this.

Yes, it would be helpful if you can produce an example of how to do an equivalent study w/ spectrumImport. 

And can I expect a future version that would overcome this limitation of the solver method? I really hope so as this would be a major part of my research.

Thank you.

dforbush2, 2022-07-07:
Hi @Aiswariak 

I have attached a function that will convert your time series to a spectrum suitable for the spectrumImport method. Note that because your provided time series is not perfectly periodic, there is spectral leakage in this Fourier transformation, so the wave elevation time series that results when you run wecSim with spectrumImport is not _exactly_ what is input to this function, but you can take care of that in pre-processing to make the result exactly correct. Testing with your time series, these results are much more in line with what was expected. After conferring with the team I will add this function to the repository.

Steps to follow:
1) run ampSpectraForWS (see comments in code for description of inputs).
2) save outputs to YourFileName.mat file, as appropriate for use with spectrumImport. Define all three columns.
http://wec-sim.github.io/WEC-Sim/master/user/code_structure.html#spectrumimport
3) change your wecSimInputFile waveClass initialization to:
```

 % Irregular Waves with imported spectrum
waves = waveClass('spectrumImport');      
waves.spectrumFile = YourFileName.mat';

```

4). Run wecSim.

dforbush2, 2022-07-07:
I am going to close this issue, as this function and workflow will solve your problem. Feel free to open another if you encounter further difficulties

closed 2022-07-07 by dforbush2

Aiswariak, 2022-07-07:
Thank you so much for the timely help and support. I will try this.