WEC-Sim/WEC-Sim #878: [Theory or Implementation]

hamzaJayy, 2022-06-03:
Hi,

I am confused how the hydrostatic restoring force is being calculated by WEC-Sim. What I understand is, if the body is partially submerged, and floats on the surface, then hydrostatic restoring force will be hydrostatic coefficient times the displacment of the body. The body height is 1.6m and it is 0.8m out of water when it is at equilibrium position. So, would not the displacement of the body  be in the rangre of -0.8 to 0.8 for calculation of the hydrostatic restoring force?

In my case, the significant wave height is 12.2m. So the body is heaving much more than the height of the body. WEC-SIM is calculating the hydrostatic force by multipying the hydrostatic coefficients by the total displacment of body CG from mean water surface level, which in return giving very large restoring force than it should be.

Since I am using linear hydro, and to staturate the restoring the force between the maximum and minimum force, I tried fixing the problem by introducing saturate block to limit the displacement in restoring_force_library_block between -0.8 to 0.8, which is the bottom and top face of body. By doing so, the heave displacment of the body becomes execissvely large i.e 2 to 3 times of the wave height. 

Can someone guide me why WEC-sim uses total dispalcement of body CG from mean surface level instead of body submergance in water to calculate to hydrostatic restoring force?

dforbush2, 2022-06-06:
Hi @hamzaJayy 

Because you are using linear hydrostatics, the force calculation is, as you have noted, based upon the displacement of the body CG, not the instantaneous free-surface. This is fairly standard practice for handling linear hydrodynamics. If your wave length is sufficiently long that the body's heave motion is in phase with the wave displacement, this is an accurate assumption.

Modeling a shorter wave length, and thus very steep and possibly breaking wave, where the above is not true and your body motion is not in phase with wave elevation, is a highly non-linear phenomena. At a minimum, non-linear hydrostatics should be used, but I would encourage you to consider if the other assumptions of linear hydrodynamics (which are used to calculate wave radiation and excitation forces as well) are appropriate for your modeling situation: 12 m is an extremely large wave field!

Finally, if the body's motion is restricted such that it will be fully submerged or out of the water in a given wave field, this sort of force saturation is non-linear, and will not be handled appropriately by linear hydrostatics.

Hope that helps.

hamzaJayy, 2022-06-09:
Hi @dforbush2 

I was trying to model WEC with mooring using nonLinearHydro in WEC-sim. I am giving the hydrodynamic coefficients calculated by NEMOH (linear BEM) as .h5 file. As, body(1).nlHydro = 2, so WEC-sim is using non-linear hydrodynamics to calculate to the restoring force. By hand calculation, I know the maximum hydrostatic restoring force is around 400kN. When nonLinearHydro is turned on, should WEC-sim cap the restoring force between that limit. To test the fidelity of nonLinearHydro simulations, 1DOF (heave) motion is being studied. 

I have run a decay test, as expected,nonLinearHydro case is caping the hydrostatic restoring force than linearHyrdo (see figure 1, decay test). 

In 2nd and 3rd figures, as you can see, the nonLinearHydro is not accurately capping the restoring force within the limits. For some sea state, it is trying to cap but some peaks are appearing. But for few states, nonLinearHydro is not even affecting the restoring force at all. 

I am confused why nonLinearHydro results are different in decay test and in different sea states. I have tried reducing the time step (simu.dtNL ) of nonLinearHydro block, but no change. Can you please guide me whats the rationale behind these results? or please refer me to some theory or anything. 

Another question is, when using nonLinearHydro, the hydrodynamic coefficients calculated using linear theory BEM can be used or they also need to be updated?

[image]
[image]
[image]

dforbush2, 2022-06-23:
This issue is related to #886 and appears to share a root-cause.  Discussion continues on that thread. This issue will be closed to avoid confusion.