WEC-Sim/WEC-Sim #879: [Feature request]

niyurui, 2022-06-07:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
How to get the power in the sag direction in rm3 case and draw a graph.

**Describe the solution you'd like**
The solution I consider is to write a program in wecSimInputFile.m

**Describe alternatives you've considered**
The alternative I consider is to write a program in userDefinedFunctions.m

**Personal project context**
I want to find out the input power, output power and efficiency of the oscillating float wave power generation device in the project I am responsible for.
**Additional context**
_Add any other context or screenshots about the feature request here._

jtgrasb, 2022-06-07:
Hi @niyurui,

Thank you for your question. The userDefinedFunctions.m file automatically runs at the end of WEC-Sim and is intended for post-processing the results of the WEC-Sim simulation, so that is the best place to set up processing and analysis scripts. 

As for the input, output power, etc., WEC-Sim outputs a structure called "output" to the workspace, which contains a variety of settings and simulation results. Within this output structure the wave elevation is recorded as well as the physical response of each body and PTO power considering the damping and stiffness specified. For example, the wave elevation and time can be accessed through the variables "output.wave.elevation" and "output.wave.time", respectively, while the output power to the PTO can be accessed through "output.ptos.powerInternalMechanics". 

I hope this is helpful and let me know if you have further questions!

niyurui, 2022-06-08:
@jtgrasb  
Hello, I have found the output. ptos.powerinternalmechanics. What I want to know is why the power here is negative. What I want to know is the power that rms can output to the outside world. In addition, I want to know how to draw power graphs in MATLAB, such as those that pop up automatically. Thank you very much.

jtgrasb, 2022-06-09:
@niyurui 
The output.ptos.powerInternalMechanics variable is defined as The net power lost in the joint DOF due to stiffness and damping [https://wec-sim.github.io/WEC-Sim/dev/user/api.html?highlight=powerinternalmechanics](url). WEC-Sim uses the notation that negative power means power is being taken out of the system, while positive power means power is being sent into the system. Because the PTO is taking power out of the WEC system, the power is defined as negative. Below is the Simulink block diagram, which may help explain the calculation.
[image]

The graphs that pop up automatically are based on the userDefinedFunctions.m file, which calls functions within our response class to plot position, forces, etc. You can plot the output.ptos.powerInternalMechanics vs. output.ptos.time variable (or any other desired plots) in the userDefinedFunctions.m file.

Best,
Jeff

niyurui, 2022-06-12:
@jtgrasb 
Thank you very much for your answer. Now I want to measure the electric power output by RM3 through PTOsim module, as stated in Advanced Features-Constraint and PTO Features, but I am a beginner in MATLAB, and I don't know much about how to build PTO-sim model (as shown in the figure below). Could you please provide me with a complete RM3 source file including PTO-sim module to help me learn? I would be grateful.
[image]

jtgrasb, 2022-06-13:
@niyurui 
I'd recommend checking out our WEC-Sim Applications repository - https://github.com/WEC-Sim/WEC-Sim_Applications. In the PTO-Sim folder of this repo, we have provided an example of the RM3 with both direct drive and hydraulic PTOs, which should help you get started with PTO-Sim in order to build your own model.

Best,
Jeff

niyurui, 2022-06-14:
@jtgrasb 
Thank you very much. Your answer helped me a lot. There are no more questions at present. You can close or keep the questions open.

niyurui, 2022-06-15:
@jtgrasb 
Hello, I have a new problem. In the case of RM3_cHydraulic_PTO, I want to draw the current, voltage and output electric power of the generator directly in the electric generator equivalent circuit block. Can this be drawn in wecsim? How to do it? Or can you provide me with the code required by the corresponding userDefinedFunctions? I would be grateful.

jleonqu, 2022-06-15:
Hello @niyurui 

The voltage and the current are in the 'output' structure after you run WEC-Sim. Once the simulation is done, you can use the next lines to have the timeseries of current and voltage:

Voltage = output.ptosim(6).voltage;
Current = output.ptosim(6).voltage;

And the you can calculate the power as the product of Voltage and Current.

niyurui, 2022-06-16:
@jleonqu 
Hello, I'm here to continue to discuss with you about wecsim. I found the output current, voltage and power of ptosim in the way you said, and drew the image as follows:
 Current：
[image]
Voltage:
[image]
Electric power:
[image]
It can be seen here that the output electric power reaches more than 15000W.
In addition, the graph of powerInternalMechanics obtained after running RM3 case in tutorials is as follows:
PowerInternalMechanics:
[image]
It can be seen here that the peak power of PTO from WEC system can reach more than 550,000 W, which is far from the electric power output from PTOsim. Therefore, I have a new question, how to find the efficiency of PTOsim, that is, the efficiency of WEC converting mechanical power into electric power. If it is convenient for you to answer, I will sincerely thank you for your help.

jtgrasb, 2022-06-24:
@niyurui 
The output.ptos.powerinternalmechanics is zero because the pto damping and stiffness values are defined as 0 in the wecSimInputFile.m. This is because the pto characteristics are defined within the ptoSim variables/blocks, which is separate from where output.ptos.powerinternalmechanics is being calculated. In order to get the PTO input power, I would recommend using P = F*v: The PTO force is output.ptosim(1).forcePTO, and the velocity is the relative velocity between the two bodies. By multiplying force times velocity, you can get the mechanical energy. 

@jleonqu Let me know if you have any other suggestions/comments since you have more insight on the ptoSim side.

niyurui, 2022-06-26:
I've got a general idea. Your answer is very enlightening to me. Thank you very much.

closed 2022-06-26 by niyurui