WEC-Sim/WEC-Sim #881: [Feature request]How to view the electric power output of the device in the RM3 example？

woosalute, 2022-06-09:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
     Dear developer, I am learning the simulation steps of RM3 recently. My final graduation thesis is also very similar to this device. My purpose is to calculate the power generation efficiency of the device, so now I want to know how to view the output electric power in the output results.

**Describe the solution you'd like**
Can you tell me how to view its output electric power? If not, how to change the Simulink model or code?

**Describe alternatives you've considered**
I found output ptos. Powerlnternalmechanics in the output folder（As shown in the following figure）, but I don't know what the power here represents and why it is negative,. I would be grateful if you could tell me in detail!

[image]

**Personal project context**
My purpose is to calculate the power generation efficiency of the device, so now I want to know how to view the output electric power in the output results.If possible, how can we directly get the efficiency of the device? Efficiency is the ratio of output power to input power

**Additional context**
N/A

jtgrasb, 2022-06-13:
Hi @sinclairwan, thanks for your questions.
I recommend checking out issue #879, which addresses many of the same questions you have here. To fully simulate a PTO system to get the output power, you can use PTO-Sim, an example for which can be found in the WEC-Sim Applications repo.

Best,
Jeff

sinclairwan, 2022-06-14:
Dear@jtgrasb, thank you very much for your guidance！ I have found the answer I want in the issue #879  you said, next I will carry out further simulation, I will communicate with you in time if there is a question, thank you again！

kmruehl, 2022-06-15:
@jtgrasb  and @sinclairwan thank you!

closed 2022-06-15 by kmruehl