WEC-Sim/WEC-Sim #883: Question about Morison Element implementation

ahmedmetin, 2022-06-12:
Hello,

I would like to use the Morison Element function as a better drag model than the currently implemented quadratic friction model that does not take into account the relative velocity between buoy and wave. For the purpose, I set the inertia terms in the Morison element to 0 and keep only the drag term. In the quadratic friction model I only use the rotational friction terms. (The flow is assumed to be irrotational).

I have also a floating cylindrical buoy with centre of gravity at the centre of the cylinder. I assume the point of application of Morison force to be at centre of gravity.

I check the Morison element drag force and observe that the drag force is not continuous and becomes 0 every time the position of the buoy is above the mean water line, even though the buoy is still inside water. 

Then I see that inside the Morison force implementation the force is set to 0 if the point of application goes above the mean water line. See screenshot. 
[image]

Why is the force set to 0? It feels that it is wrong at first sight, but of course I do not understand the reason. I think the Morison force should be set to 0 only if the buoy goes completely out of water. Therefore, one should check the position of the buoy with respect to the local wave elevation. And it feels better to check the centre of each mesh if it is under water, rather than if the point of application is under water. One can also maybe add a correction for the characteristic area depending on how much percent of the buoy is submerged. What do you think?
Thanks! 

Best regards,
Ahmed

nathanmtom, 2022-06-15:
@ahmedmetin Thank you for your question; however, we require that any issues follow the issues template that we have made to ensure we receive the appropriate information to recreate your issue. Please update your issue above to follow the template, see Issue #881 as an example, or open a new issue. Once you have completed this the team can begin assisting with your issue. 

Cheers, 
Nathan
National Renewable Energy Laboratory

ahmedmetin, 2022-06-15:
@nathanmtom Thanks for your response. I have tried to put it in the template now. I am not sure if this classifies as a feature request, bug or just a question.

ahmedmetin, 2022-06-15:
Thank you so much for the thorough explanation, @nathanmtom. I did not realize before that I can enter more than one Morison element. I will definitely try and see how it works. Thanks a lot! :)

Best regards,
Ahmed

ahmedmetin, 2022-06-19:
Hello @nathanmtom, I have tried the Morison element force with more more Morison elements. However,  I get a difference between the case with more Morison elements and single Morison element where I remove the condition z > 0. 
Currently, I run a very simple case to compare the two cases. Just a decay test in heave. With more Morison elements I get a much more damped response. The reason is that the characteristic area is constant for each Morison element in the axial direction (heave direction), because of constant water plane area of cylinder. So when one has more Morison elements, the total characteristic area increases with the number of Morison elements. I have the same drag coefficient in both cases, i.e. the drag coefficients of the Morison elements with the case with multiple Morison elements is equal to the drag coefficient of the case with single Morison element. I guess the answer should be they should not be equal, at least for the axial case to have a matching response, right? 

Thanks! 

Best regards,
Ahmed

ahmedmetin, 2022-06-19:
Update. It seems I need to divide the area by the number of Morison elements, because the total characteristic area basically increases if I do not do that.
Please see the figure below. I get basically convergence with only 4 elements as you suggested.

[image]

Thanks a lot for your help. I think we can  close the issue. 
Best regards,
Ahmed

closed 2022-06-19 by ahmedmetin