WEC-Sim/WEC-Sim #886: Issue in Nonlinear Buoyancy and Froude-Krylov Wave Excitation Tutorial - Heaving Ellipsoid

hamzaJayy, 2022-06-15:
Dear Developers,

I have doubts in simulation results of WEC-sim application of non-linear hydrodynamics for a Heaving Ellipsoid.

I ran simulation using the settting and files provided by WEC-sim application master for a heaving ellipsoid for 4 different cases, which are given below:

1. Linear hydrodynamics with PTO
2. Linear hydrodynamics without PTO
3. Non linear hydrodynamics with PTO
4. Non linear hyrdrodynamics without PTO

Here, PTO is a simple damper. Without PTO means making the damping coefficient zero in wecsim input file. I was expecting that wecsim will calculate the response of the ellipsoid using non-linear hydrodynamics when the body(1).nlHydro =2 no matter there is a PTO or no PTO. But this is not the case. The system behaves linearly even the body(1).nlHydro =2 with PTO  turned off (as it can be seen in graphs below)

All other settings kept constant. Regular wave with h=12.2m and T=16.61 is simulated with simu.wave('regular').

Is there something I am missing about nonLinearHydro formulation using Froude-kryolov force? Please guide me what is happening. 

I have attached the output graphs of Heave, restoring force and force exciatation for you reference in the same order below: 
[image]

[image]

[image]

**Workstation Information (this section MUST BE completed):**
 - OS: , Windows 10
 - MATLAB/Simulink Release, 2021b
 - WEC-Sim Release v4.4

I am not sure where is the issue. But it does not make sense to me for now. I am facing another issue with nonLinearHydro that I asked in #878 . I guess this issue is related to theory implementation.

dforbush2, 2022-06-21:
Hi @hamzaJayy 

Thanks for bringing this to our attention. This is a bug: the nonlinear hydro implementation is not properly initializing for some reason in all cases. Note that the shown results are not in absence of a PTO, but with pto.c = 0 . If the PTO block is removed and the system is thus fixed, everything does appear to be working properly.

This is likely the root cause of the issue you present in #878 , though I do encourage you to update to v5.0 when available, as the fix will be applied to the current repository. 

We will prioritize this fix.

hamzaJayy, 2022-06-23:
@dforbush2 

Thank you for response. Can you please look at the case #878 . In that case there is no PTO, only buoy attached to a catenary mooring line. I am only changing the sea states. The restoring force becomes linear from nonlinear at certain sea states. Is nonLinear hydro sensitive to sea states?

Further, any idea when v5.0 will be available?

dforbush2, 2022-06-23:
V5.0 is available now. I am also considering your results to #878. The problem seems to be related to the non-linear hydro implementation itself, however it is not straightforward. We are working to solve this problem. The most concerning feature, as you point out, is the peaks in excess of the body's maximum displacement force, and that the presence of restoring force saturation seems to be affected by the presence of a PTO. We are prioritizing this fix.

dforbush2, 2022-07-08:
Update: Not yet sure what causes the problem, but looking at body.geometry.center (these are the center of the stl file nodes: remember that the COG is at [0 0 -2], which is the origin of the body centered coordinate system while the model is running). 
Panel 1101 is the lowest panel of the model, 
Panel 357 is the highest
Panels [946,1076,1169,1290,2370,2466, 2611] are  <1 cm off the equator in z.

Based on my runs so far, I believe the issue to be a reference frame issue between the instantaneous wave elevation calculation and the panel position (given relative to CG) that results in the calculation not correctly identifying which panels are in the water or when the body is fully submerged. 

Still working on confirming that this is the case and fixing.  

Note that this calculation still calculates pressure on the panel based on the submergence depth relative to the still water line, not the instantaneous wave height. This was intended, but a full picture of the exerted forces when nonlinear hydro is used must also include Froude-Krylov components.

dforbush2, 2022-07-12:
Hi all,
So I think I have this figured out: it is working as expected, but this should not probably be the way it is working.

Here is a snapshot of the pressure calculation, from "nonLinearBuoyancy.m", lines 32 through 37 

```
% Zero out regions above the mean free surface
z=center(:,3); z((z-elv)>0)=0;

% Calculate the hydrostatic pressure at each triangle center
pressureVect = rho*g.*[-z -z -z].*-av;
p = rho*g.*-z; 
```
Here "center" is the N x 3 matrix describing the location of the center of each STL panel at this instant in time. "av" is the normal vector of each panel at each instant in time, and "elv" is the free surface elevation relative to still water at the X,Y location of each STL panel. 

Here, we zero out z if it is about the _actual water elevation_ at that panel location. But, you'll note that p (and pressureVect) are calculated not from the difference (z - elv) but from z, which is the panel position _relative to the still water line_.

In very large waves when the device is free to move, this creates a scenario where, depending on how in-phase the motion of the body is with the wave, there are moments where the body is neither fully submerged nor fully out of the water, but is sufficiently far from the free-surface that the calculation on line 37 exceeds the saturated value. This is the reason for the transients exceeding saturation values before the flats in your trace labeled "nonlinear", while the flats themselves represent the saturation case where the body is fully immersed or in the air.

The reason that this does NOT occur with the "NLWithoutPTO" case is that the PTO damping is set to zero, and this changes the phase of the body relative to the wave, so the "transients" actually persist throughout the wave cycle.

As you have shown, this implementation is appropriate for bodies free in heave subjected to waves significantly larger than the vertical dimension of the body. 

Hopefully this answers your question with respect to the results you are seeing! Thank you for bringing this to our attention. To be clear, this is not physically correct. I will leave this issue open for the time being and discuss with the team the appropriate course of action.

salhus, 2022-07-13:
> CAUTION: This email originated from outside of NREL. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi Dom,

That was a great catch !

Sal

kmruehl, 2022-08-03:
Close this issue once #910 is merged

dforbush2, 2023-03-21:
I am reopening this issue because the PR associated with it #910 introduces a bug into the non-linear hydrodynamics calculation that results from a misunderstanding originating here. I will try to re-address the user questions here with an improved understanding. 

Bottom line up front: the implementation prior to #910 was correct and should be reverted. However, functionality with nonLinearHydro == 1 should be reconsidered as a result. It is recommended that the nonLinearBuoyancy.m implementation that is recommended in #910 be used for nonLinearHydro == 1, which is appropriate only in quiescent (i.e., no wave) sea-states.

dforbush2, 2023-03-21:
In quick summary: the Froude-Krylov calculation consists of two terms that contribute to the pressure on a body panel, each of which is calculated in WEC-Sim by a different function. Firstly, in "nonLinearBuoyancy.m" a (mostly) _linear_ hydrostatic term $$\rho g z$$ (Note especially this is z of the panel center _and is not adjusted by instantaneous surface height in any way_), where z is the position relative to the SWL of the _wetted_ panels. A wetted panel is determined by a logic check: z-elv > 0: if this is not true, the panel does not contribute any pressure onto the body. This is calculated in "nonLinearBuoyancy.m" and the total over the body is reported in the output structure as output.bodies.forceRestoring. 

There is a confusing implication here that was at the root of the erroneous #910 and the answer to this issue: In the event of a very long wave and high wave and a wave-following floating body, at the wave crest many wetted panels will have a large +z value. This will imply that the reported forceRestoring of the body can (potentially greatly) exceed the mass * gravity force of the body. This is physically counterintuitive, but its non-physical dynamic effect is (in theory, numerically it is imperfect but very close) canceled out by the contribution of the nonlinear second term. 

Secondly, in "nonFKForce.m" the pressure field exerted by the displaced free surface is calculated over _all_ the (wheeler stretched) panels. Notably, this calculation is performed twice: once presuming the wave elevation and body displacement are both zero (i.e., linear Froude Krylov) and again including the instantaneous free surface and position. The difference is strictly the non-linear contribution, since the linear FK forces are already accounted for in the BEM excitation coefficients.  The total of this calculation over the body is summed with the linear excitation coefficients and the result is shown in 'forceExcitation' (which is _oppositely_ signed with forceRestoration). 

As a result, a more intuitive presentation of these results is to plot the difference forceExcitation - forceRestoration fields of the output structure which will consider both constituent terms of the non-linear froude krylov force calculation. 

Finally, in the case posted above with PTO.c = 0, the body is largely acting as a wave follower: because of this, we should not expect a significant difference between the linear (nonLinearHydro == 0) solution and the non-linear solution. To see the effects of non-linearities more clearly we can look to the pto.c ~= 0 case, in which there is sufficient phase shift between the body and the free surface that it is no longer acting as a wave follower.

kmruehl, 2023-03-22:
@kmruehl review this

nathanmtom, 2023-05-05:
We are closing this issue and flagging PR #1017 to revert back the nonlinear hydrostatic and Froude-Krylov implementation.

closed 2023-05-05 by nathanmtom