WEC-Sim/WEC-Sim #887: [Theory or Implementation] Moments of Inertia input and BEM->WEC-Sim implementation

Dadidal, 2022-06-16:
**Is your question request related to a problem? Please describe.**
I do not understand:
1. My floating body is ship model (scale 1:50) that consists of 2 parts - Hull of 21.4 kg and additional heavy part of 10 kg inside Hull (see Figure). The draft in such configuration is 40 mm. The draft of Hull without Heavy part is 30 mm.
[image]
Therefore Center of Mass and Center of Volume of my floating body are not the same. What configuration should I calculate in BEM solver (Capytaine) for further import to WEC-Sim:
 - Hull mesh with CoG, mass and draft of Hull+Heavy part (31.4 kg and 40 mm)? And further nothing add to WEC-Sim
 - or Hull mesh with CoG, mass and draft of Hull only (21.4 kg and 30 mm)? And further in WEC-Sim add Non hydrodynamic body for Heavy part.
 - or Hull outer surface with CoG (CoV) and mass of Hull but with draft of Hull+Heavy part (21.4 and 40 mm)? And further in WEC-Sim add Non hydrodynamic body for Heavy part.

2. What values of Moments of Inertia have to be used in WEC-Sim input file? In documentation there is Ixx Iyy and Izz moments of inertia and therefore I have to insert moments related to X,Y and Z global axis, right? What should I do if I have non-zero Ixy Iyz and Ixz moments of inertia? In other words when I analyzed my floating body I found that axes of Principal Moments of Inertia does not correspond to Global frame axes.

**Describe the theory or implementation approach that you have a question**
I use Capytaine BEM solver and then import BEM results to WEC-SIm using BEMIO

**Personal project context**
I try to model crane ship in time domaine. My crane ship consists of several parts. In WEC-Sim I want to create model using Non Hydrodynamic bodies for above water parts. Therefore in BEM solver (Capytaine) I analyze only submerged part. Since mass, Center of Gravity and draft of submerged part ONLY are not the same as mass, CoG and draft of FULL SHIP I want to understand what combination of Mesh+Mass+CoG+Draft I should calculate in BEM solver to create correct model in WEC-Sim.

kmruehl, 2022-06-29:
Related to https://github.com/WEC-Sim/WEC-Sim/issues/859

Dadidal, 2022-06-30:
@kmruehl Thank you for your answer! But I still have following misunderstandings:
1. If I should model the case "Hull mesh with CoG, mass and draft of Hull+Heavy part (31.4 kg and 40 mm)? And further nothing add to WEC-Sim" what is the point then to use Non hydrodynamic bodies? From your answer I see that frequency-domain response will change with changing the CoG on the same draft. Anyway if we will add any non hydrodynamic body with specific mass to WEC-Sim model CoG will be changed and therefore the model will no have sence.
2. using `doc bodyClass` command I checked the help for WEC-Sim and found that bodyClass.inertia property is

> (`3x1 float vector`) Rotational inertia or mass moment of inertia [kg*m^{2}]. Defined by the user in the following format [Ixx Iyy Izz]. Default = ``[]``. 
But you told that definition in WEC-Sim is 3x3 matrix. Where is the truth  - is it 3x3 matrix or 3x1 vector? If it is 3x1 vector my previous question is still open.

kmruehl, 2022-07-05:
@Dadidal 

1. These are actually two separate questions. I'll attempt to respond to both:
   - a) For the BEM run in Capytaine, you'll want to cut the ship's geometry at the still water line for the overall system (ship and point mass near the bow), and mesh the resulting wetted surface area. Running Capytaine with the origin at the cg of the overall system will give you the BEM hydrodynamic coefficients at hydrostatic equilibrium. For the WEC-Sim run, you'll model the ship as one hydrodynamic body, using the mass and moment of inertia for the overall system (ship and point mass).
   - b) You are correct. You _could_ model the overall system (ship and point mass) as a combination of a hydrodynamic body (for the ship), and a non-hydrodynamic body (for the point mass). However, this approach would require you to run Capytaine for the ship only (with a shallower draft and smaller wetted surface area). This approach is not ideal because the BEM hydrodynamic coefficients would not be for the overall system at hydrostatic equilibrium. As soon as you add the point mass in WEC-Sim, the wetted surface area would increase, but you would be using BEM data for the ship only. If the difference in displacement is small, this may be a reasonable approach. However, I would recommend 1a over this approach. The intended role of a non-hydrodynamic body is to model a body that is fixed or above the SWL. Examples include the [OSWEC](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/OSWEC) and  [WECCCOMP](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/WECCCOMP).  
2. My apologies, thanks for point this out. You are correct, inertia is only modeled in WEC-Sim as a [3x1] in a body-fixed frame about the center of gravity.

kmruehl, 2022-07-19:
@Dadidal does this resolve your issue?

closed 2022-07-27 by kmruehl

Dadidal, 2022-07-28:
@kmruehl Thank you for your answer, but I still not clear about second question:
> 2. My apologies, thanks for point this out. You are correct, inertia is only modeled in WEC-Sim as a [3x1] in a body-fixed frame about the center of gravity.
Therefore what Moments of Inertia I should use:
- principal moment of inertia Ip1, Ip2 and Ip3 (axes of them p1 p2 p3 are not the same as axes X Y and Z)
- moments of inertia Ixx, Iyy, Izz? In that case axes are the same as X Y and Z, but I have non-zero Ixy, Ixz, Iyz and, while WEC-Sim uses only [3x1] vector