WEC-Sim/WEC-Sim #889: [WEC-Sim Applications] Why the output matrix value is 0

sinclairwan, 2022-06-21:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
E:\WEC-Sim\WEC-Sim_Applications-master\PTO-Sim\RM3\RM3_cHydraulic_PTO

**Describe the bug or issue**
I followed the tutorial and ran the instance successfully.But when I open forceInternalMechanics and powerInternalMechanics in output.pto, the values in these two files are all 0, since I am a beginner of wecSim, I can't understand the reason for this result well, Can you tell me why both of their matrices are 0? Why are these two matrices not 0 when the PTO-sim module is not added (as shown in $WEC-Sim\tutorials\RM3), I look forward to your reply!
**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to 'E:\WEC-Sim\WEC-Sim_Applications-master\PTO-Sim\RM3\RM3_cHydraulic_PTO'_ and run it.
1. Double click the 'output' and add it to workspace_
1. _choose 'output.ptos.powerlnternalMechanics' and 'output.ptos.forcelnternalMechanics'_
1. _See error: they are all 0_

**Expected behavior**
I hope you can explain this result

**Screenshots**
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release] 2021a
 - [WEC-Sim Release]v5.0

**Additional context**
Using wecSim is part of my graduate thesis work, your answer is very important to me, thank you!

dforbush2, 2022-06-21:
See #447 and #510 
In short, the internalMechanics field returns force or power related only to the pto.k and pto.c parameters contained within the pto blocks. When an external actuation force is prescribed, like in the example you are using, the PTO forces are defined externally, and pto.k and pto.c are zero.

sinclairwan, 2022-06-22:
@dforbush2 
Thank you for your reply! I am very happy that you have noticed my problems.
After thinking about your answer,I can understand why the values are 0.
Now i want to calculate the efficiency of the RM3 with hydraulic PTO(the example in WEC-Sim_Application/PTO-Sim/RM3), I have calculated the output power based on the current and voltage output of the generator, but when calculating the input power, I chose to multiply output.pto.velocity and output.ptosim(1).forcePTO as the input power ,and then divided the output power by the input power to obtain instantaneous efficiency. I am not sure whether this method right or not,  but it seems that the power efficiency is around 30%.
 Can you tell me what is the efficiency of the device in general, so that I can judge whether my calculations are reasonable.
I am looking forward to your answer! Thank you again!
[image]
[image]

dforbush2, 2022-06-22:
I would caution you against using instantaneous efficiencies. I would instead calculate on the basis of average power for a given sea state. So for electrical power, I would integrate the product of voltage and current over your run, and divide that by the integral of the product of pto force and pto velocity. This is a more robust approach to efficiency calculation that will extend well to advanced controllers and sea states. 

Note that this is NOT equivalent to taking the average of your instantaneous efficiencies.

I'm tagging @jleonqu if he has any specific insights on this case!

sinclairwan, 2022-06-23:
Hi! @dforbush2 
Thank you very much for your answer， it helped me a lot!
I try to understand the calculation method you have provided in your comment. Can I understand it this way, the efficiency of the device can be calculated according to the formula shown in the figure below?
[image]
Since the data is discrete, I turned the integral into a summation.
I am very much looking forward to your reply！

sinclairwan, 2022-06-24:
Hello！@dforbush2
Thank you for your patience in answering my questions! You are so kind!
 My current design is to integrate three RM3 units together and form a floating energy platform together with the offshore wind turbine. Is the efficiency of these three RM3 units together the same as the efficiency of a single unit. If not, how can  i calculate the efficiency?
[image]

dforbush2, 2022-06-24:
That is a more complicated question. Because the wave field seen by each device is not identical, I would not assume that they have identical efficiencies. The formulae above should work for each individual device, but unless they share a common PTO it would not be appropriate to apply these to the group. While the wave fields are not identical, they are exciting the same frequency bands, so I would expect the individual device efficiencies to be close.

Assuming the schematic you have attached is to-scale, it will be important to run your BEM code with all three RM3 devices modeled, and run WEC-Sim with body-to-body interactions turned on. 

Because your initial questions were answered, I am going to close this issue. If you have further questions please feel free to open another issue. This way it is easier for other users to find the answers they need.

closed 2022-06-24 by dforbush2