WEC-Sim/WEC-Sim #899: [BUG] Images for PTO-Sim blocks not showing on Library Browser

jleonqu, 2022-06-29:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
_A clear and concise description of what the perceived bug is._
The PTO-Sim blocks have some images to help the users to understand what kind of component is modeled in the block. The images are not visible in the Library Browser, but they are shown when the blocks are moved to the simulink model.

**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._
No specific files needed, just the WEC-Sim source code.

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to Library Browser in Simulink
2. _Scroll down to PTO-Sim library and select the Hydraulic library. This is the current view in the Library Browser:
[image]

3. _Drag one of the blocks to Simulink. This is the current view in Simulink:
[image]

4. _The image is visible in Simulink, but not in the library browser

**Expected behavior**
_A clear and concise description of what you expected to happen._

The images should be visible in both the Library Browser and Simulink.

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

Screenshots were provided above.

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) 2021b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v5.0

**Additional context**
_Add any other context about the problem here._

jleonqu, 2022-06-29:
This issue was solved in PR #896

kmruehl, 2022-06-29:
@jleonqu I'm going to expand this issue to refer to the larger WEC-Sim Library issue. @dforbush2 tagging you for awareness

dforbush2, 2022-06-29:
I had some luck getting library blocks to appear by:
1). addWecSimSource.m (if not in path already)
2). Opening blank simulink model and opening library.
3). Clicking this arrow in your screenshot to "Enable HDL Compatible Mode"
[image]

4). After this loads up, in the command line type
`hdllib off`

This seems to do the equivalent of refreshing the library, but I've restarted my machine and matlab and I haven't had to do this again, whereas previously I would need to re-load the library each time. I actually kind of liked this for switching back and forth between lib versions, forks, etc. so buyer beware on this fix.

(I still get the ??? graphic on the PTO-Sim blocks themselves but I think this is to be fixed by PR #896 ?)