WEC-Sim/WEC-Sim #905: [WEC-Sim Applications]

AmirRahimi70, 2022-07-06:
**Describe the bug or issue**
_First of all, I would like to appreciate WEC Sim Team to develop codes. About my issue when I run free-decay simulation I catch an error:

WEC-Sim Input From Standard wecSimInputFile.m Of Case Directory... 
WEC-Sim: An open-source code for simulating wave energy converters
Version: 5.0

Initializing the Simulation Class...
	Case Dir: C:\Users\amir\Documents\MATLAB\WEC-Sim-5.0\WEC-Sim_Applications-master\Free_Decay\1m 
Elapsed time is 0.537883 seconds.

WEC-Sim Pre-processing ...   
Warning: Removed
'C:\Users\amir\Documents\MATLAB\WEC-Sim-5.0\WEC-Sim_Applications-master\Free_Decay\1m\output'
from the MATLAB path for this MATLAB session.
	See 'doc path' for more information. 
> In simulationClass/setup (line 162)
In initializeWecSim (line 208)
In run (line 91)
In wecSim (line 43) 
	Infinite water depth specified in BEM and "waves.waterDepth" not specified in input file.
Set water depth to 200m for visualization.
Elapsed time is 0.126590 seconds.

WEC-Sim Simulation Settings:
	Start Time                     (sec) = 0
	End Time                       (sec) = 40
	Time Step Size                 (sec) = 0.01
	Ramp Function Time             (sec) = 100
	 Number of Time Steps     = 4000 

Wave Environment: 
	Wave Type                            = No Wave (Convolution Integral Calculation)

List of Body: Number of Bodies = 1 

	***** Body Number 1, Name: sphere *****
	Body CG                          (m) = [0,0,-2]
	Body Mass                       (kg) = 261724 
	Body Diagonal MOI              (kgm2)= [2.09073E+07,2.13061E+07,3.70855E+07]

List of PTO(s): No PTO in the system

List of Constraint(s): Number of Constraints = 1 

	***** Constraint Name: Constraint1 *****

Simulating the WEC device defined in the SimMechanics model C:\Users\amir\Documents\MATLAB\WEC-Sim-5.0\WEC-Sim_Applications-master\Free_Decay\1m\sphere.slx...   
Elapsed time is 0.490174 seconds.
Elapsed time is 8.457525 seconds.

Post-processing and saving...   
Error using wecSim (line 45)
Error evaluating 'StopFcn' callback of Frames block (mask) 'sphere/Global Reference Frame'.
Callback string is 'run('stopWecSim')'
Caused by:
    Error using wecSim (line 45)
    Index exceeds the number of array elements. Index must not exceed 1._
Would you please help me what to do to fix this error.
Please inform me if you need more info.
Thank you in advance,
Amir

**Workstation Information :**
 - OS: Windows 11
 - [MATLAB/Simulink Release 2021a
 - [WEC-Sim Release  v4.5 and v5.0

salhus, 2022-07-13:
Hi @AmirRahimi70 ,

I tried running the 1 m case in the applications repository. I was not able to recreate your issue. May you please upload  your working folder as a zipped file.

My suspicion is that it is being caused by a userdefinedfunctions.m file, where the variables to plot are not correctly defined. Looking at the outputs you pasted, the issue is triggered in the post-processing stage.

Cheers,
Sal

AmirRahimi70, 2022-07-13:
Thank you Sal ([salhus](https://github.com/salhus)),
Yes the problem was related to the "userdefinedfunctions.m" file. I changed the file in a correct way and then the error was fixed.

salhus, 2022-07-14:
@AmirRahimi70 
Sounds like that resolved the issue. I noticed that you reopened the issue. Did you identify any other issues? if not, can this issue be closed?

AmirRahimi70, 2022-07-15:
Yes the issue was resolved. I reopened the issue so that you can add your comments if you want.
best wishes,
Amir

closed 2022-07-15 by salhus