WEC-Sim/WEC-Sim #908: [WEC-Sim Applications]Visualization with ParaView

AmirRahimi70, 2022-07-10:
Hello 
I installed paraview as well as all required packages to run the WEC-Sim.py file. I added WEC-Sim.py to paraview macros. But When I entered a .pvd file and clicked on WEC-Sim macro, I faced following error in paraview output message log:

Traceback (most recent call last):
  File "<string>", line 8, in <module>
  File "C:\Program Files\ParaView 5.10.1-Windows-Python3.9-msvc2017-AMD64\bin\Lib\site-packages\paraview\simple.py", line 763, in Hide
    raise ValueError ("proxy argument cannot be None when no active source is present.")
ValueError: proxy argument cannot be None when no active source is present.
Traceback (most recent call last):
  File "<string>", line 8, in <module>
  File "C:\Program Files\ParaView 5.10.1-Windows-Python3.9-msvc2017-AMD64\bin\Lib\site-packages\paraview\simple.py", line 763, in Hide
    raise ValueError ("proxy argument cannot be None when no active source is present.")
ValueError: proxy argument cannot be None when no active source is present. 
**Workstation Information (this section MUST BE completed):**
Would you please help me to fix that error.
Thank you in advance,
Amir

 - OS:  Windows 11
 - MATLAB/Simulink Release  2021b
 - WEC-Sim Release V5.0
 - ParaView v5.10.0 and v5.10.1

**Additional context**
_Add any other context about the problem here._

kmruehl, 2022-07-13:
@AmirRahimi70 are you having this issue with one of the WEC-Sim_Application Paraview cases, or is this a different case? If it's the latter, can you please ZIP the model and attach it to this issue? Thank you.

kmruehl, 2022-07-13:
@AmirRahimi70 I don't have the same configuration as you, but I was able to run the https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Paraview_Visualization/OSWEC_NonLinear_Viz case, using the WEC-Sim documentation instructions for [Loading a ParaView State File](https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#loading-a-paraview-state-file) without issue. 

The error you're receiving seems to be due to a proxy configuration. 

- OS: Windows 10
- MATLAB/Simulink Release 2021a
- WEC-Sim Release V5.0
- ParaView v5.9.1

AmirRahimi70, 2022-07-15:
> @AmirRahimi70 are you having this issue with one of the WEC-Sim_Application Paraview cases, or is this a different case? If it's the latter, can you please ZIP the model and attach it to this issue? Thank you.

Hello, I have this issue with both of the WEC-Sim_Application Paraview cases. I can see cases in paraview but I am not able to run the .py macro files. 
[image]

kmruehl, 2022-07-19:
@AmirRahimi70 this is likely a Paraview version compatibility issue. We updated to Paraview 5.9.1 last year, and we have not updated since. I recommend downgrading your Paraview installation to 5.9.1, that's the [version WEC-Sim currently supports](http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#install-paraview-and-wec-sim-macros).

kmruehl, 2022-07-27:
@AmirRahimi70 have you tried Paraview 5.9.1? Did that resolve your issue?

closed 2022-08-03 by kmruehl

AmirRahimi70, 2022-10-21:
Yes, I tried Paraview 5.9.1 and resolved the problem.