WEC-Sim/WEC-Sim #912: [WEC-Sim Applications] clarifying how to use MoorDyn

mohammadadibzadeh, 2022-07-16:
Dear all,

I have modelled a floating WEC with 5 bodies. It has a central body that 4 pre-tensioned mooring lines must moor. I have been trying to build my moorDyn file like the [RM3](https://wec-sim.github.io/WEC-Sim/master/user/tutorials.html#user-tutorials-rm3) geometry coupled with [MoorDyn](https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#mooring-moordyn) but I could not able to realize how to change the file for my case to run without errors. 
The water depth in my model is 3 m, and the central body draft (height of body submerged in water is 1.5 m; therefore, the unstretched mooring length would be 1.5 m. To make a pretension, I would define the mooring length as 1.3m. (The connection points of mooring at their top and end should be like this: connection point to the central body: [0.5, 0, -1.5] and connection point to the sea bed [0.5, 0, -3]).
I am sharing my mooring file here. I would be grateful if any could help me with my problem.
[lines.txt](https://github.com/WEC-Sim/WEC-Sim/files/9126569/lines.txt)

dav-og, 2022-07-27:
Hi @mohammadadibzadeh , I just wanted to confirm if you've followed the instructions for installing MoorDyn as explained here: https://github.com/WEC-Sim/MoorDyn

If so, and you're still seeing errors, please share your model input files (ideally as a minimum working example) so that we can try to reproduce the error.

Also, were you able to run the [RM3 MoorDyn example case from our applications repo](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Mooring/MoorDyn) without modification? (i.e. was it only after changing some inputs that you saw some errors?) Or was it not running at all?

Cheers,
Dave

kmruehl, 2022-08-03:
This is related to #914

mohammadadibzadeh, 2022-08-03:
Dear Dave,

Thank you for getting back to me. 

I've followed the instructions for installing MoorDyn, and I can run the [RM3 MoorDyn example case](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Mooring/MoorDyn) without modification. The problem is that I encountered errors while modifying it to run my own WEC. As I mentioned, To make a pretension, I would define the mooring length as 1.3m. (The connection points of mooring at their top and end should be like this: connection point to the central body: [0.5, 0, -1.5] and connection point to the sea bed [0.5, 0, -3]).
I am sharing all files here.

Thanks in advance for your help.

Kind regards,

Mohammad
[MBFWEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/9253060/MBFWEC.zip)
a

dav-og, 2022-08-10:
Hi @mohammadadibzadeh,

I've ran your model and I'm seeing the following output from MoorDyn:

[image]

It seems like the 'CB' option you've used in the lines.txt input file isn't recognized:
[image]

I'm not sure what you're trying to do, but it seems like this needs to be changed to one of the following options:
[image]

**(“Fixed”, “Vessel”, or “Connect”)**

Please check out the MoorDyn documentation: https://moordyn.readthedocs.io/en/latest/usage.html#connection-properties and if you have further issues with MoorDyn inputs then please reach out to the MoorDyn team.

Cheers,
Dave

mohammadadibzadeh, 2022-08-19:
Hi @dav-og 
Thank you for answering my question.
As you mentioned, the problem was defining the name of the floating object; hence the CB should be changed to Vessel and Fix to Fixed. I have run the model successfully.

Kind regards,
Mohammad

dav-og, 2022-08-22:
Excellent - thanks for letting us know, Mohammad.

closed 2022-08-22 by dav-og