WEC-Sim/WEC-Sim #913: [Feature request] Update Capytaine BEMIO examples to v1.4

kmruehl, 2022-07-18:
**Is your feature request related to a problem? Please describe.**
N/A

**Describe the solution you'd like**
[Capytaine v1.4](https://github.com/mancellin/capytaine/releases/tag/v1.4) was recently released. WEC-Sim's examples are based on a prior version of Capytaine, and should be updated to v1.4. This will remove the dependency on Meshmagick for hydrostatic calculations. 

**Describe alternatives you've considered**
N/A

**Personal project context**
N/A

**Additional context**
Updating the WEC-Sim examples to Capytaine v1.4 should eliminate some of the prior package dependencies and ease Capytaine installation for users.

kmruehl, 2023-01-04:
Updated issue to reflect latest Capytaine release. @dav-og  and @salhus  are working with @mancellin  to update Capytaine to support multi-body, and then WEC-Sim examples

dav-og, 2023-01-26:
just opened a Capytaine PR to include a Hydrostatics.dat and KH.dat file writer: https://github.com/capytaine/capytaine/pull/285

once this is approved we can update the Capytaine BEMIO examples 👍

closed 2023-05-17 by kmruehl