WEC-Sim/WEC-Sim #914: [Theory or Implementation] Mooring Matrix

mohammadadibzadeh, 2022-07-19:
Hi 
Hope you are doing well!
I have some question about modeling a floating multibody WEC with 4 pre-tensioned mooring lines?
My WEC model has 5 bodies, a central body which is moored by 4 pre-tensioned mooring lines to sea bed and 4 floating bodies that hinged to central body. Here, I modeled moorings using mooring matrix and moorDyn but I think none of them works correctly. So, I am sharing my wecSimInputFile and .slx model image and I would be grateful if someone could help me to modify mooring matrix. Distant between top and bottom connection points of moorings are 2m, their Unstretched length must be 1.985m, and stiffness are 137482N . For example, line 1 should connect to body in [0.5, 0, -1] and connect to sea bed in [0.5, 0, -3]; line 2 should connect to body in [0, 0.5, -1] and connect to sea bed in [0, 0.5, -3] and so on. 

Thanks in advance for taking your time.
[wecSimInputFile1.txt](https://github.com/WEC-Sim/WEC-Sim/files/9140822/wecSimInputFile1.txt)
[image]

dav-og, 2022-07-27:
Hello @mohammadadibzadeh, please could you share all of your other input files? (i.e. .stl, .h5 & .slx files)
Cheers,
Dave

dav-og, 2022-07-28:
Hi @mohammadadibzadeh if you could also fill in the issue template we created that would be helpful:
[image]

mohammadadibzadeh, 2022-08-03:
Thank you for your reply.
I have attached all the needed files

[MBFWEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/9253112/MBFWEC.zip)

dav-og, 2022-08-10:
Hi @mohammadadibzadeh,

I ran your model and saw the following error:

[image]

After running BEMIO the model seems to run fine for me:

[image]

The simulation runs for 100s without error:

[image]

What is concerning is that in still water the bodies 'pop' up - indicating that your system is not in hydrostatic equilibrium for the mass and position variables you've defined in your input file. This likely means that your hydrodynamics will be off - as they were computed for the initial positions at t=0s, whereas actually you system ends up oscillating about a higher position (~ +0.3m). In short, when you run still water simulations, nothing should move.

Please let me know if this resolves your issue and if we can close this out 👍

Cheers,
Dave

dav-og, 2022-08-17:
Does this answer your issue?

mohammadadibzadeh, 2022-08-19:
Dear @dav-og 

Thank you for taking your time and for the great guidance you provided.
Your opinion about the issue of hydrostatic equilibrium was correct and needed to be solved. However, I could run the model correctly after some try and error.

Thanks again,
Mohammad

dav-og, 2022-08-22:
Glad to hear you've resolved your issue! 👍

closed 2022-08-22 by dav-og