WEC-Sim/WEC-Sim #915: [Theory or Implementation]

BenVickers98, 2022-07-21:
Hi there.

This may seem like a silly question, but I'm very new to hydrodynamic modeling.

I am trying to model a heaving body in a wave flume, with narrow walls which leave minimal space between the body and the wall surfaces.
I am trying to model this system in 2D, or as close to 2D as I can approximate. 
When applying the hydrodynamic coefficients, RAOs, and IRF in WEC-Sim is it possible to simply model the heave displacement over time, without the need for information on the other degrees of freedom.
Moreover, does the time-domain solver solve the relevant equations individually for each degree of freedom, and hence would not need any additional coefficients for the other degrees of freedom?

Thanks for any help on this. Many apoligies if my question is not clear.

nathanmtom, 2022-07-27:
@BenVickers98  Thank you for submitting your issue. Before we are able to answer your question we ask that you edit the existing issue to follow one of our issue templates. An example of our template is shown below for reference.

[image]

dav-og, 2022-07-27:
Hi @BenVickers98 

> in WEC-Sim is it possible to simply model the heave displacement over time, without the need for information on the other degrees of freedom.

You can do this by applying a linear (prismatic) joint on the body in the vertical direction. This will effectively eliminate the other DoFs in the system so that your floating body only moves in heave.

You would still need to run the BEM code for 6 dofs though, as this how WEC-Sim is set up to run with Simscape Multibody.

**Beware:** Simscape Multibody only permits us to modify 9 out of the 36 elements of a rigid body's mass matrix (a mass scalar and a 3x3 inertia tensor). Hence, we combine all of the linear added mass terms and add them to the body's scalar mass term. This generally works ok, but when we look at heave-only systems we end up with too much added mass.

Please refer to L8 in this file: https://github.com/WEC-Sim/WEC-Sim_Applications/blob/1e097e1396100824440883d54e506115e045456a/Free_Decay/1m/wecSimInputFile.m#L8

for how to use simu.adjMassFactor to "back out" the correct added mass term for heave.

dav-og, 2022-08-10:
Hi @BenVickers98, I'll close this out for now but let us know if you need more advice 👍

BenVickers98, 2022-08-14:
Hi, thanks for getting back to me.

That's great news! 
My only other question on this matter is to do with why the specific value is used:

simu.adjMassFactor = 0.47761120825;

I would be very grateful for some clarification of how the specific value for this adjusted Mass factor is obtained and whether it needs to vary depending on the geometry of the body etc.

Thanks for the help so far.

Kind regards,
Ben

BenVickers98, 2022-08-14:
Hi David,

Yes I have just submitted an additional related question in response. However the issue is currently closed.
Would it be possible for the issue to be reopened and to get a reply to my question.

Thanks a lot for the help so far.

Kind regards,
Ben

dav-og, 2022-08-17:
Hi Ben,

WEC-Sim sums the 3 on-diagonal linear added mass terms (surge-surge, sway-sway, heave-heave) together, and adds one term to the body's mass scalar variable.

Hence, if you only allow a system to move in heave, the added mass is too much.

Therefore, for these kinds of decay tests, the summed added mass term (surge-surge + sway-sway + heave-heave) is multiplied by simu.adjMassFactor to 'back out' the added mass term of interest.

In short, for the sphere example I shared : (AM11 + AM22 + AM33)*0.47761120825 = AM33

You would have to determine the correct simu.adjMassFactor value for your particular case.

Cheers,
Dave

BenVickers98, 2022-08-18:
Hi David,

Thank you for getting back to me so quickly.

My final question on this matter is for which heave-heave, sway-sway, and surge-surge added mass coefficients do I obtain this value, in terms of which frequency component? 

When calculating: AM33/(AM11 + AM22 + AM33) to obtain simu.adjMassFactor, it varied significantly for each frequency.

Should I be using a value for when the frequency is infinite? i.e  A(∞)

Thanks again for all the help,

Kind regards,
Ben

dav-og, 2022-08-24:
Hi @BenVickers98 , sorry I overlooked this - yes you're right - you should use the A(∞) values 👍

nathanmtom, 2022-09-13:
@BenVickers98 The WEC-Sim wanted to check if you had any further questions regarding this issue? If all of your current questions have been answered we would like to close this issue.

Cheers, 
Nathan 
National Renewable Energy Laboratory

nathanmtom, 2022-09-20:
@BenVickers98 Since we have not heard from you since August 18th, we are going to close this issue and assume you have a solution to your initial inquiry. 

Cheers, 
Nathan
National Renewable Energy Laboratory

closed 2022-09-20 by nathanmtom