WEC-Sim/WEC-Sim #916: [BUG] Elevation import does not work with rampTime = 0

jtgrasb, 2022-07-21:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
The elevation import method for the waveClass gives the following error when the rampTime equals zero.

[image]

**WEC-Sim file(s)**
Seems to be related to the way the elevation file is read in within the waveClass

**To Reproduce**
_Steps to reproduce the behavior:_
1. Go to RM3 example
2. Change waves to elevation import and rampTime to 0.
3. Run wecSim.

**Expected behavior**
WEC-Sim run without error.

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - MATLAB 2021b
 - WEC-Sim v5.0

**Additional context**

kmruehl, 2022-08-03:
Close this issue once #917  is merged

nathanmtom, 2022-08-17:
Closing this issue as #917 has been merged into the the `dev` branch of WEC-Sim.

closed 2022-08-17 by nathanmtom