WEC-Sim/WEC-Sim #918: [BUG] Unable to resolve the name 'clock_out.time' (Desalination)

H0R5E, 2022-07-25:
**Describe the bug**

There is a CI failure for the desalination application case e.g. [here](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/2707787456).

**WEC-Sim file(s)**

Seems to be related to [this line](https://github.com/WEC-Sim/WEC-Sim/blob/fceb0a8318465b1b0bcdf02d9b13b3b5ae9bc22a/source/functions/postProcessWecSim.m#L7) in `postProcessWecSim.m`. I guess the global `clock_out` or it's field `time` is not being set in this case.

Here is the trace:

```
================================================================================
  Error occurred in TestDesalination/testDesalination and it did not run to completion.
      ---------
      Error ID:
      ---------
      'MATLAB:undefinedVarOrClass'
      --------------
      Error Details:
      --------------
      Unable to resolve the name 'clock_out.time'.
  
      Error in postProcessWecSim (line 7)
      simu.time = clock_out.time;
  
      Error in stopWecSim (line 21)
      postProcessWecSim
  
      Error in run (line [91](https://github.com/WEC-Sim/WEC-Sim_Applications/runs/7438361219?check_suite_focus=true#step:10:93))
      evalin('caller', strcat(script, ';'));
  
      Error in wecSim (line 48)
      run('stopWecSim');
  
      Error in TestDesalination/testDesalination (line 54)
                  wecSim
  ================================================================================
```

**Additional context**

This [commit](https://github.com/WEC-Sim/WEC-Sim_Applications/commit/56851c578c49e8190e25576033331943379bc959) was made by @kmruehl in an attempt to correct the issue, but it didn't work.

kmruehl, 2022-07-27:
@H0R5E I've been working with the desal case lately, so I'll work on this BUG. I should be resolved with #923

kmruehl, 2022-07-27:
@H0R5E I've tested this locally on both 2020b and 2022a (the two MATLAB versions we're running CI for the applications repository), and both passed. I'm unable to reproduce this error locally. 

Nonetheless, we tried a few things to debug, none of which resolved the issue:
- @jtgrasb resolved the bug with the WEC-Sim Library in #923, by saving the WEC-Sim Library to 2020b. 
- In https://github.com/WEC-Sim/WEC-Sim_Applications/pull/28 I updated the broken library links (one needs to remain broken, but the library block was updated to the latest version on WEC-Sim dev before I broke it again), and I saved the desalination model file (`OSWEC_RO.slx`) to 2020b (which shouldn't be necessary, but was worth trying). None of these resolved the issue. 

Do you have any other ideas why it would be failing remotely, but passing locatlly? And why is this case passing on WEC-Sim master?

kmruehl, 2022-07-27:
> Following #920, I also checked the SLX file versions in WEC-Sim_Applications and there are 13 that are too modern, including the Desalination case:
> 
>     * `Cable\MBARI_cable.slx`
> 
>     * `Desalination\OSWEC_RO.slx`
> 
>     * `Mean_Drift\WECSim\sphere.slx`
> 
>     * `Mooring\MooringMatrix\RM3MooringMatrix.slx`
> 
>     * `RM3\RM3_DD_PTO\RM3_DD_PTO.slx`
> 
>     * `Passive_Yaw\PassiveYawRegression\OSWEC.slx`
> 
>     * `Traditional_Morison_Element\hydroBody\monopile.slx`
> 
>     * `Traditional_Morison_Element\morisonElement\monopile.slx`
> 
>     * `WECCCOMP\WECCCOMP\WaveStar.slx`
> 
>     * `WECCCOMP\WECCCOMP_Fault_Implementation\WaveStar.slx`
> 
>     * `WECCCOMP\WECCCOMP_Nonlinear_Model_Predictive\WaveStar.slx`
> 
>     * `Wave_Markers\OSWEC\OSWEC.slx`
> 
>     * `Wave_Markers\RM3\RM3.slx`
> 
> 
> I'm not sure it's relevant to this bug, but it does stop me opening the files properly.

@H0R5E this is related to my [prior comment](https://github.com/WEC-Sim/WEC-Sim/pull/919#issuecomment-1197219534), WEC-Sim model files (i.e. case files) do not need to be saved to 2020b. Only the WEC-Sim Library needs to be saved to 2020b. There are occasionally issues when a case file is saved to a newer version of MATLAB, but it's not usually an issue. All of the other applications test are passing, despite being saved in newer versions of MATLAB.

H0R5E, 2022-07-28:
Kelley, I can't run it locally myself because I don't have Simscape Fluids. Have you been testing using Linux? The runners are all Linux VMs.

H0R5E, 2022-07-28:
@kmruehl, from the [actions logs](https://github.com/WEC-Sim/WEC-Sim_Applications/runs/7115646175?check_suite_focus=true) I identified [the commit](https://github.com/WEC-Sim/WEC-Sim/commit/68278c3549a0be2c29ad5ee81b82c1d665dca211) in WEC-Sim where the issue started. I think this is the identified commit because it starts calling `stopWecSim` rather than introducing a bug.

jtgrasb, 2022-07-28:
@kmruehl @H0R5E I just looked into this issue. It seems that for some reason the .slx model for the desal case is simply missing the clock_out variable, which should be output to workspace in the global frames block. Here's our RM3 model:
[image]
And here's the OSWEC_RO model:
[image]

I think it should just be a simple fix to add the clock_out to workspace block to the .slx model.

kmruehl, 2022-07-28:
@jtgrasb I thought that was the issue too, but that's one of the revisions I made on https://github.com/WEC-Sim/WEC-Sim_Applications/pull/28, and it didn't resolve the issue. I updated the global reference frame block by replacing the block with the latest version from the WEC-Sim Library, and keeping the library link. Feel free to look into the issue and see if you have any other ideas.

kmruehl, 2022-07-28:
> Kelley, I can't run it locally myself because I don't have Simscape Fluids. Have you been testing using Linux? The runners are all Linux VMs.

I haven't run it on LINUX, and don't have a good way to

jtgrasb, 2022-07-28:
> @kmruehl, from the [actions logs](https://github.com/WEC-Sim/WEC-Sim_Applications/runs/7115646175?check_suite_focus=true) I identified [the commit](https://github.com/WEC-Sim/WEC-Sim/commit/68278c3549a0be2c29ad5ee81b82c1d665dca211) in WEC-Sim where the issue started. I think this is because it starts calling `stopWecSim` rather than introducing a bug.

@H0R5E Could you clarify what you mean by it starts calling `stopWecSim` rather than introducing a bug? I don't quite understand what you're getting at.

--Thanks,
Jeff

kmruehl, 2022-08-24:
@H0R5E can you clarify?

H0R5E, 2022-08-24:
Hi, so the commit I identified puts `run('stopWecSim')` into `wecSim.m`. Because the desalination case had an unlinked version of the Frames block, I'm surmising that `stopWecSim` wasn't being called at all in this case, as the version of the global reference frame was older than the change that added calling `stopWecSim` inside it.

So basically, we can't identify the commit that broke this example because the code in `stopWecSim` has not been called since #548, I reckon.

H0R5E, 2022-09-01:
Kelley, I can't do anything to the models because I don't have Simscape Fluids. Can you try switching on "Single simulation output" in the Model Settings -> Data Import/Export tab? Screenshot below:

[image]

H0R5E, 2022-09-01:
Kelley, I can't really help much because I don't have a license for Simscape Fluids.

A sticking plaster would be to mark the test as an expected failure for now.

kmruehl, 2023-05-17:
I'm able to run this test locally without issues. Since this issue has been open for a while, and we've unable to resolve the issue on the runner, I'm going to go ahead and remove the Desalination test from the runner. We can still test it locally, but at least it'll resolve the failed test issue on our repo.

kmruehl, 2023-05-17:
The desal test was remove in https://github.com/WEC-Sim/WEC-Sim_Applications/pull/28

closed 2023-05-17 by kmruehl