WEC-Sim/WEC-Sim #920: [BUG] SLX files with versions incompatible with R2020b

H0R5E, 2022-07-25:
**Describe the bug**

I'm currently running R2020b and I noticed that some SLX files will not open because they are saved in a more modern MATLAB version. I have created [some new tests](https://github.com/WEC-Sim/WEC-Sim/pull/919) to indentify SLX files that have versions which exceed 2020b.

**WEC-Sim file(s)**

The list of affected files is:

- `examples/OSWEC/OSWEC.slx`
- `examples/RM3/RM3.slx`
- `examples/RM3FromSimulink/RM3FromSimulink.slx`
- `source/lib/WEC-Sim/WECSim_Lib_Body_Elements.slx`
- `source/lib/WEC-Sim/WECSim_Lib_Frames.slx model`

**Additional context**

Because my MATLAB is R2020b I can't convert the SLX files back to the correct version. If someone could save them in the correct format, that would be great.

kmruehl, 2022-07-27:
Should be resolved with #923 by @jtgrasb

closed 2022-07-27 by kmruehl

h7ossein, 2023-02-25:
Hello @kmruehl , So I have this problem, and I'm quite confused about what I should do to resolve this without downgrading my MATLAB. I'm using OSWEC example in particular (if that's important). I would really appreciate your help.

jtgrasb, 2023-02-27:
Hi @h7ossein, thank you for your question. Please open a new issue using the one of the templates to indicate the MATLAB version, specific files, etc. you are using. This will help us answer your question quicker.

PR #923 (2022-07-27): Save slx files in 2020b fixes #920
This PR resolves #920.
commit bc9208e4 (2022-07-27): Save slx files in 2020b fixes #920 (#923)

* Add files via upload

* Add files via upload