WEC-Sim/WEC-Sim #921: [WEC-Sim Applications]

POGULURI, 2022-07-27:
Hello, we recently started using WEC-Sim to simulate pitch-type WEC motion (Salter's duck) where the motion response is measured based on fixed position that is center of rotation (COR). In addition, we have thoroughly examined the WEC motion response in the past using WAMIT, Orcaflex, and Star-CCM+, and all software matches well. As we are planning to use WEC-Sim these days, there are a few issues that we have encountered while using WEC-Sim that I have explained below. For comparison, we also simulated with Orcaflex using a linear solution with no nonlinear terms incorporated or activated.

Note: When I use the word center of gravity COG or COR I mean the below
w.r.t (COG) means - All hydrodynamic coefficients are given in terms of COG as well as the body moment of inertia (MOI)
w.r.t COR means - All hydrodynamic coefficients are given in terms of COR as well as the body moment of inertia (MOI)

Main issue::
1. The simulations were carried out with two different wave heights (H = 0.1m and 2.0m) in relation to COG and COR using WEC-Sim and Orcaflex, and the response curves were compared. Please see the attachment. We forcefully changed the data read in BEMIO (readWAMIT) in WEC-Sim to read data from COR rather than COG.
We noticed that regardless of wave height H = 0.1m, the WEC body response to COG and COR matches well, which is supported by Orcaflex results, but it is not the same with H = 2.0 m. There are differences all around the peak.

Could you tell me what made the difference in the results? Because these simulations are only performed by changing the COG to COR. Please help us.

Thank you in advance. 
[image]
[image]

dav-og, 2022-07-27:
Hi, thanks for your question.

I'm just wondering if you could provide some additional detail about your model, particularly what is referred to by "external damping"?
[image]

These kinds of RAO discrepancies can commonly be observed for different wave heights (e.g. Figure 22, Babarit et al (2011) - The NumWEC Project):
[image]
https://tethys-engineering.pnnl.gov/publications/numwec-project-numerical-estimation-energy-delivery-selection-wave-energy-converters

Its typically due to velocity-dependent terms providing additional damping - and becomes most pronounced around the resonance frequency - exactly as you've shown. Hence, I think this could just be related to your 'external damping' force?

dav-og, 2022-07-27:
I've just realized that you are actually seeing different RAOs for WEC-Sim CoG and CoR, both in 2m waves.

I'm hesitant to say if this could be a bug or not without some additional information about the models. Please could you provide all input files for the two models, as well as additional information about:

> We forcefully changed the data read in BEMIO (readWAMIT) in WEC-Sim to read data from COR rather than COG.

POGULURI, 2022-07-27:
> Hi, thanks for your question.
>
> I'm just wondering if you could provide some additional detail about your model, particularly what is referred to by "external damping"? [image]
>
> These kinds of RAO [...]

Thank you so much for responding.
This could have been possible if the simulations were run with nonlinear terms in the equation of motion. In our case that should not have happened because we performed solely on a linear basis. As you mentioned later, the results obtained with COR and COG for H= 2.0m vary whereas they were the same for H=0.1m.

I can provide the input files if needed.

dav-og, 2022-07-27:
Please can you try comparing 0.1m and 2.0m with no viscous damping (to see if this is causing the discrepancy, as in the NumWEC example), and if the error persists please open up a new issue using the bug report template, with some additional details about your modification to readWAMIT() 👍
Cheers,
Dave

closed 2022-07-28 by dav-og