WEC-Sim/WEC-Sim #926: [Theory or Implementation]

hamzaJayy, 2022-08-01:
**Is your question request related to a problem? Please describe.**
I am trying to model a floating body using nhBody=2 (drag body) with morison element. While modelling the element, I found that analysis is sensitive to body.dispVol. After certain value of dispVol, the MorisonForce becomes significatly high resulting in very large motion of body. I have attached the model parameters and the following error.

[image]

the error:
[image]

the input wave conditions are
[image]

As you can see, the diamater of body is 0.2m. If I make it 0.1, the model runs fine without any error. But by increasing the diameter (hence increasing dispVol), singularity in motion comes in. The problem vanishes if I increase mass of the body while incresing the diameter. 

**Describe the theory or implementation approach that you have a question**
My current understanding of morison element is that system should not become unstable even if body is neutrally buoyant. but thid is not the case with WECsim implementation for now. 

This is reponse of body when the diameter is 0.1m
[image]

As Morison force = inertial + drag

hand calculation shows there will be sigficant drag force as compared to inertial force, but the graph shows the dominant one is inertial as it follows the position of body. (I am not really sure about this) 

**Personal project context**
I am trying to model oscilation of submerged cylinder when held in place by taut mooring line. Mooring line is being modelled uising MoorDyn. Above system despriction does not have mooring line, it is just a cylinderical body submerged at 25m in 50m sea, and constrainted to move in surge only. 

I tried to model the body motions with mooring line present as well. The system becomes unstable (or shows singularity) when dispVol is increased. 

**Describe the type of conclusion or resolution you are looking for**
I am seeking guidance why moriosn element is so sensitive to the choice of dispVol. Also please looking into dragForce calculation for Morison element

dforbush2, 2022-08-01:
Hi @hamzaJayy 

Can you please provide a *zip of your run so I can look directly?

My instinct is that this is related to the hydrostatics force. Because you are defining the mass and displacement volume explicitly in you input file, you may be creating a situation where the body is far from neutrally buoyant. This can create an instability in the hydrostatics force, especially if resisted by the "taut cable" you described, leading to large velocities. These large velocities will also lead to very large Morison Element forces, but these are likely a symptom of the underlying hydrostatic instability rather than the cause.

hamzaJayy, 2022-08-02:
[FilesToRun.zip](https://github.com/WEC-Sim/WEC-Sim/files/9238426/FilesToRun.zip)

Hi @dforbush2 

I attached the files above. At the moment, both mooring initialization in wecSimInputFile and mooring block in simulink have been commented out. 

1) Try running it with diameterOfCon =0.1. it will run. Then try changing it to 0.2. it will not. (without mooring)

2) Then the same with mooring line. Simulation will run with diameterOfCon=0.1 and it wont when diameterOfCon=0.2. Pretty strange.

Just to clarify, mooring line is a taut with both end fixed (anchored). First anchor is at sea level and the other end is at seabed. It is a model resembling a concentrated mass in a taut string and excited by sea waves.

hamzaJayy, 2022-08-03:
Hi @dforbush2 

Previously, I was using v4.4. Now, I shifted to v5.0. I have attached the new files for you to run. 

The schematic I am trying to model is as follows: 

[image]

I have observed the following things:

1) when the mooring lines are not present, and body(1).mass = 'equilibrium' is used with fixed constraint, the simulation runs fine and calculate the morison force. But, error comes up, saying that the velocity of the body is infinite when the constraint is changed from 'fixed' to 'translational' (surge direction). 

2) when the mooring lines are included as shown in picture above, with body(1).mass = 'equilibrium' is turned on, the simulation does not run, saying that there is sigularity in velocity (constraint is "translation"). Its ineresting to see that the taut mooring lines are not restricting the body when its velocity is shooting out of roof. 

3) I found out that If the mass of body is increased by some factor (>>1)  than the displaced mass of water (body.volume*rho), the simulation runs fine with 'translation' constraints. Seems like, if the mass of body is not high enough to limit the acceleration of the body, the simulation shows sigularities in velocities. 

4) The mooring line tension does not change if I used very heavy mass at the middle (refer to schematic above). 

[freshFiles.zip](https://github.com/WEC-Sim/WEC-Sim/files/9248541/freshFiles.zip)

hamzaJayy, 2022-08-08:
Hi @dforbush2 ,

I am trying to dig into the issue. I would like to update about my current progress in findind the root cause of the problem. I guess I am near it, not sure though.

I made the problem simple. So, body is simply floating on surface as body(1).mass = 'equilibrium'. The wave condition is 'noWaveCIC'. The body cg is at [0,0,0], cb=[0,0,-2], initial.displacement = [0,0,-1]. I removed the mooring lines too.
[updatedFiles.zip](https://github.com/WEC-Sim/WEC-Sim/files/9278645/updatedFiles.zip)

1) Firstly, I used nonHydro = 1, the simulation runs fine, as results as expected. body is not moving and stays there. 

2) The, I used nonHydro = 2, with morisonElement=0; the simulation runs fine, as body does nothing (as expected).

3) now the funny part. I used the nonHydro =2, with morisonElement =1 (and separately tried  morisonElement=2, as well). I get the following error again, 

[image]

then I checked the output. The results are as follows: 

[image]

As you can see, the acceleration is Nan, and the same case is with morison element. Since the velocity of body is zero at time = 0.0, the derivative of velocity is getting out of whack (is it so?). Then why not deriviative of displacement not going inifinate as well? 

@dforbush2 @kmruehl can you please guys suggest me how to further dig into the problem. I really need to get this done soon. Your timely response will highly appreciated. Thanks.

I am attaching updatedFiles for you guys.

nathanmtom, 2022-08-11:
@hamzaJayy Thank you for continuing to look into the issue and provide our team additional information. I'm stepping in for @dforbush2  and after reviewing your code I believe the issue is that you have defined `simu.rampTime=0` as we have identified an issue with our ramp functions causing `NaN` when we divide by `simu.rampTime=0`, please see the current PR #917 describing the issue and intended fixes. 

When I modified your `simu.rampTime=10`, WEC-Sim ran to completion getting you past your error listed above. However, I cannot comment on if the simulations are performing as desired. 

If we have answered your question about getting past the error at time = 0 then we would like you to close this issue. If you have other questions or issues with your model we ask that you open a new issue. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

nathanmtom, 2022-08-17:
@hamzaJayy I wanted to check to see if the adjustment to the `simu.rampTime` solved your issue? If I do not hear back from you by the end of the week, I'll close this issue but you can reopen if you still have questions. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2022-08-19:
@hamzaJayy Since we have not heard from you I am going to assume your initial issue was resolved and will close this issue.

closed 2022-08-19 by nathanmtom