WEC-Sim/WEC-Sim #927: [Theory or Implementation] BEMIO\NEMOH\RM3

usamatoman, 2022-08-02:
I am trying to regenerate the same results in WEC-Sim\examples\BEMIO\NEMOH\RM3. 
I run axiMesh.m for both the float and spar separately to generate Hydrostatics.dat, KH.dat, RadiationCoefficients.tec, and ExcitationForce.tec for each body. 
Then, I rename the two Hydrostatics.dat files to Hydrostatics_0.dat and Hydrostatics_1.dat for the float and spar, respectively. And the same for KH.dat files. However, I can't combine the two RadiationCoefficients.tec files (for both bodies) into one file that BEMIO requires as an input. Would you please help me with that?

dav-og, 2022-08-03:
Hello @usamatoman,

The hydrostatics calculation treats each body in isolation - resulting in the two .dat files you referred to.

The BEM solution treats this as a two body problem (see [the Nemoh input file](https://github.com/WEC-Sim/WEC-Sim/blob/master/examples/BEMIO/NEMOH/RM3/Nemoh.cal)). Hence, any BEM outputs from Nemoh (e.g. RadiationCoefficients.tec, ExcitationForce.tec) will contain all of the data for 2 bodies (including hydrodynamic interaction terms).

On a side note, Nemoh was rewritten recently as [Capytaine](https://github.com/capytaine/capytaine) and we're encouraging people to use this open-source BEM code instead of Nemoh 👍. If you have any issues using Capytaine, please open an issue on that repository 🙂.

Cheers,
Dave

closed 2022-08-03 by dav-og

usamatoman, 2022-08-03:
Thank you so much @dav-og