WEC-Sim/WEC-Sim #928: Programming language

aglassFHKiel, 2022-08-03:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
_Identify which project this issue is associated with._

**Describe the issue**
_A clear and concise description of the issue being encountered and any necessary context._

Hello,

which Programming language use wec sim?

best regards
Andreas

dav-og, 2022-08-03:
hello @aglassFHKiel 

I'm not sure I fully understand your question, but WEC-Sim is written in MATLAB/Simulink and uses Simscape Multibody. For more high-level information about WEC-Sim please [check out our documentation](https://wec-sim.github.io/WEC-Sim/master/index.html).

closed 2022-08-03 by dav-og

aglassFHKiel, 2022-08-03:
Hello David,

i would like to know whether wec sim is written in C++, Python or another language?

I am not really sure if MATLAB is an own Programming language

Mit freundlichen Grüßen
Andreas Glaß
--------------------------------------------------------
[FuE_25-Jahre_Aktionslogo_RGB_E-Mail]
FuE-Zentrum FH Kiel GmbH
M.Eng. Andreas Glaß
- Wissenschaftlicher Mitarbeiter -
Schwentinestr. 24
D-24149 Kiel
Germany

Fon         +49 431 210 2593

E-mail      ***@***.******@***.***>
Web        www.fh-kiel-gmbh.de<http://www.fh-kiel-gmbh.de/>
Handelsregister: Amtsgericht Kiel, HRB 4312
Vertretungsberechtigter Geschäftsführer: Björn Lehmann-Matthaei
Prokurist: Christian Wantrupp
--------------------------------------------------------
Die in dieser Nachricht enthaltenen Informationen sind vertraulich und/oder rechtlich geschützt. Sie sind nur für den Adressaten bestimmt. Bitte benachrichtigen Sie den Absender, falls Sie nicht der beabsichtigte Empfänger sein sollten, und löschen Sie diese Nachricht umgehend aus Ihrem System. Danke.

This e-mail may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorised copying, disclosure or distribution of the material in this e-mail is strictly forbidden. Thank you.

dav-og, 2022-08-03:
Yes, MATLAB is a programming language 👍
Cheers,
Dave