WEC-Sim/WEC-Sim #932: [Theory or Implementation]

niyurui, 2022-08-09:
**Is your question request related to a problem? Please describe.**
I don't understand how the components of hydraulic PTOsimulink model work in harmony in RM3 case with hydraulic PTO, that is, the working principle of hydraulic PTO.

**Describe the theory or implementation approach that you have a question**
Working principle of hydraulic PTO

**Personal project context**
I tried to describe the working principle of hydraulic PTO to readers in my paper.

**Describe the type of conclusion or resolution you are looking for**
Figure shows the simlink simulation model of hydraulic PTO used in RM3 case. What puzzles me is how the rectifier valve, high-pressure accumulator, low-pressure accumulator and hydraulic motor work together to drive the rotary generator to generate electricity. Could you please tell me the working principle of the hydraulic PTOsimlink model, or if there is any relevant text description for reference? I would be very grateful for your opinion.

**Additiona
[image]
l context**

dav-og, 2022-08-10:
Hi @niyurui,

I would recommend reading So's original PTO-Sim paper: https://energy.sandia.gov/wp-content/uploads/2014/06/2015-IEEE-PES_PTO-Sim_Nak.pdf

And Yu's paper on a hydraulic PTO model for RM3: https://www.nrel.gov/docs/fy19osti/71078.pdf

Cheers,
Dave

jleonqu, 2022-08-17:
Hello @niyurui 

Does this answer your question?

closed 2022-08-24 by kmruehl