WEC-Sim/WEC-Sim #933: Morrison Element Implementation for OSWEC

vizzvanath, 2022-08-13:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
Hi! I'm trying to model viscous effects into my oswec design using Morrison element approach in which I have broken up the flap into several segments. I need to know if my implementation and understanding are correct.

**Describe the theory or implementation approach that you have a question**
The design is identical to the B-OF device given in https://www.researchgate.net/publication/242025031_Numerical_benchmarking_study_of_a_selection_of_Wave_Energy_Converters. Width of the flap is 26m, height is 12m and thickness is 2m. I want to break up the flap into 12 segments (red squares - 1mx2m) and apply the Morrison elements to the middle of each segment (blue circles). Ony the drag in the x direction is considered and the drag coefficient is chosen as 8. The COG is located 5.75m below the top (green circle). 

**Personal project context**
I'm trying to obtain similar results to what's given in the publication above.

**Describe the type of conclusion or resolution you are looking for**

Could you tell me if the code below does what I explained earlier? For body(1).morisonElement.rgME I considered the COG of the flap as the reference.

body(1).morisonElement.option = 1;
body(1).morisonElement.cd = zeros(12,3);
body(1).morisonElement.cd(:,1) = 8;
body(1).morisonElement.ca = zeros(12,3);
body(1).morisonElement.characteristicArea = zeros(12,3);
body(1).morisonElement.characteristicArea(:,1) = 26*1;
body(1).morisonElement.VME  = zeros(12,1);
body(1).morisonElement.rgME = [0 0 -5.75; 0 0 -4.75; 0 0 -3.75; 0 0 -2.75;...
    0 0 -1.75; 0 0 -0.75; 0 0 0.25; 0 0 1.25; 0 0 2.25; 0 0 3.25; 0 0 4.25; 0 0 5.25]; 

Also, what is the purpose of body(1).morisonElement.VME? I understand that it is the characteristic volume of Morrison element although having a hard time understanding how to use it. Why does it set to zero in the Desalination application? 

Thank you!

**Additional context**
[image]

nathanmtom, 2022-08-17:
@vizzvanath Thank you for your question and I will do my best to answer your question. 

First, in terms of understanding what ` body(1).morisonElement.VME` and the other parameters used to define Morison Elements, please review (if you have not already) our developer manual on the Morison Element Implementation: 

http://wec-sim.github.io/WEC-Sim/master/developer/advanced_features.html#morison-element

You are correct that the ` body(1).morisonElement.VME` variable indicates the characteristic volume of the Morison Element and is required if you want to capture forces on element due to the relative acceleration between the element and fluid velocity. By setting this variable to zero, as in the desalination case, the Morison Element will only provide a drag force related to the relative velocity squared. 

Without having run your model, reviewing how you set up the Morison Element array looks appropriate to reproduce the case you have cited.

Cheers, 
Nathan
National Renewable Energy Laboratory

vizzvanath, 2022-08-20:
@nathanmtom, appreciate your responses! I have attached my files.  Please let me know if you discover any issues.

Thank you!
[B_OF.zip](https://github.com/WEC-Sim/WEC-Sim/files/9387248/B_OF.zip)

nathanmtom, 2022-08-22:
@vizzvanath What is the version of WEC-Sim that you are running? When I've opened your files you appear to be using an older version of WEC-Sim (prior to v5.0). Based on the v5.0 version you will need to update the variable labeling if you plan on using your files. If you are using a prior version of WEC-Sim them you may not need to make changes. However, I'd suggest trying to run the code yourself and open another issue if you are still having difficulty with the implementation. Unless you have other questions about implementation and variable definition, we can close this issue and wait until you have results from simulations. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2022-08-22:
@vizzvanath The quickest way is to look at the `simulationClass.m` file and there should be a variable `wsVersion` which indicates which version of WEC-Sim you are using. Also, when you run WEC-Sim the initial print to screen text should include the version number being run in the simulation. 

Thanks for the confirmation that your model is running in your version. I am going to close this issue and invite you to open a new issue regarding questions with implementation and results. 

Cheers, 
Nathan
National Renewable Energy Laboratory

closed 2022-08-22 by nathanmtom