WEC-Sim/WEC-Sim #934: [Theory or Implementation]

BenVickers98, 2022-08-16:
I have been getting the following error when I run wecSim on my case:

"
Error using wecSim (line 45)
['HEAVE_BOAT/Global Reference Frame/Solver Configuration']: Error evaluating
equations at time 0.0.  Stopping simulation.  There may be a singularity in the
solution.  If not, try reducing the step size (either by reducing the fixed step
size or by tightening the error tolerances)
Caused by:
    Error using wecSim (line 45)
    ['HEAVE_BOAT/Translational/CONSTRAINT (Prismatic Joint)']:
    'HEAVE_BOAT/Translational/CONSTRAINT (Prismatic Joint)' has a degenerate mass
    distribution on its follower side.
 
"

I think the issue could likely come from how I created my ".h5" file.

When running the Bemio.m scripts used for the examples, I simply copied them for my model and changed the file path.

%%
hydro = struct();

hydro = readNEMOH(hydro,'/Users/benvickers/Documents/MSc Fluid Mechanics/Research Project/Time_Dom_SIMS/WEC_SIM/WEC-Sim-5.0/examples/BEMIO/NEMOH/HEAVE_BOAT');

hydro = radiationIRF(hydro,60,[],[],[],1.9);
hydro = radiationIRFSS(hydro,[],[]);
hydro = excitationIRF(hydro,157,[],[],[],1.9);
writeBEMIOH5(hydro)
plotBEMIO(hydro)
%%

The values within the radiationIRF function, radiationIRFSS, and excitationIRF are just what was already used for the other cases.

I can see that each of the examples has different values for these, so my question is what exactly do these numbers describe, and could this be related to the error that I mentioned above?

Additionally, when I ran the BEM tool, Nemoh, the radiation coefficients, diffraction and excitation forces that I get were calculated within the range of 0.5 to 8 rad/s. Is this a suitable range for inputting into WEC-Sim, or could it cause problems?

Thanks for any help on this matter.

Kind regards,
Ben Vickers

BenVickers98, 2022-08-16:
I previously submitted  [Theory or Implementation] #915 , which was closed however, I would like to reopen it in order to get some answers to my questions.

This was what I am proposing to do in WecSim:
"Restrain the motion of a floating object, so that it can only move in heave"

The response:
You can do this by applying a linear (prismatic) joint to the body in the vertical direction. This will effectively eliminate the other DoFs in the system so that your floating body only moves in heave.

I have tried to do this, however, I am slightly concerned that I have not set my Simulink model correctly.
I was unable to upload an image of my model here, so I will try to describe it.

-A global reference frame, connected to constraint(1)
-Constraint(1): Translational Constraint, which is then connected to pto(1)
-pto(1): Translational PTO, which is then connected to body(1)
-body(1): Rigid Body (My heave-only floating body)

If there is any help or useful examples that you are aware of for creating a heave-only floating structure model, and any information on whether my BEM frequency ranges are suitable and what the inputs on the bemio.m code represent, I would be very appreciative.

Kind regards,
Ben

dav-og, 2022-08-17:
It does seem like the error could be caused by how you've set up your Simulink model. The values you've passed to BEMIO are probably not suitable for this system either...(please refer to the function descriptions for more information about the arguments: https://github.com/WEC-Sim/WEC-Sim/blob/master/source/functions/BEMIO/radiationIRF.m)

Please could you fill in an issue template so that we can have a bit more information?

[image]

If you can attach your WEC-Sim model as a .zip file that would be helpful too.

Cheers,
Dave

dav-og, 2022-08-24:
Hi @BenVickers98 , did you manage to figure out the error?

nathanmtom, 2022-09-13:
@BenVickers98 The WEC-Sim team wanted to reach out to see if you were able to resolve the issues related to this post. If we do not hear from you by the end of this week we will close this issue. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

nathanmtom, 2022-09-20:
@BenVickers98 Since we have not heard from you since August 16th we are going to close the issue and assume you have found a solution to your initial inquiry. 

Cheers, 
Nathan
National Renewable Energy Laboratory

closed 2022-09-20 by nathanmtom