WEC-Sim/WEC-Sim #935: [WEC-Sim Applications]

timmaflan, 2022-08-16:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._
Hello, I am working on a point absorber similar to the RM3. I am trying to implement two mooring systems, one which attaches one mooring line to the end of the spar (one line system), and one which attaches three lines to the float (three line system). In total this would be a four line system using different types of lines for the float and the spar (stretchy v.s. rigid).  

**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._
Is it possible to model two mooring systems attached at different points with WEC-Sim, moordyn or the mooring matrix? 

**To Reproduce**
_Steps to reproduce the behavior:_
I have tried the following options: 
1. I have tried to implement a moordyn system (attached to the float) and a mooring matrix system (attached to the spar) at the two locations in the wecSimInput file, and have added both a moordyn block and mooring matrix blcok to the simulink file. This did not work, even if I set the mooring matrix to zeros, and use a moordyn file that worked previously 
2. I have tried to implement a moordyn system that records all four lines, however this does not align with the simulink file and unfortunately crashes the program
3. I have not tried strictly mooring matrix options
4. I have not tried to recreate the stiffness in another way (e.g using a PTO system)

**Expected behavior**
_A clear and concise description of what you expected to happen._
I hope to find a way to have multiple mooring systems modeled at the same time, but attached at different attachment points, however I am not sure if this is possible with the current source code. Down the line, I would also like to model a 6 line system with the lines attaching at similar points on the float

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

dav-og, 2022-08-17:
Hello, thanks for filling in the template.

It should be possible to model the type of mooring system you've described with MoorDyn. However, its difficult to know whats happening without seeing your input files - would you mind uploading your WEC-Sim model as a .zip file?

Cheers,
Dave

timmaflan, 2022-08-19:
Hi Dave, I realized I had to rename the moordyn and mooring matrix blocks in simulink so that they were different (E.g Mooring(1) and tether(1)). Otherwise, the numbering system didn't align properly, and threw errors. Thanks, Timma!

closed 2022-08-19 by timmaflan