WEC-Sim/WEC-Sim #936: [Theory or Implementation]

sinclairwan, 2022-08-19:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
    In the RM3 tutorial, the H5 file is generated by using the calculation results of WAMIT. In my own process, I tried to use the AQWA module in ANSYSWorkbench for calculation. There are a few questions I would like to confirm with you during use.
   
**Describe the theory or implementation approach that you have a question**
   1. As shown in Figure 1, the AQWA module in Workbench is divided into frequency domain analysis and time domain analysis. When using AQWA for calculation, frequency domain analysis is always performed first before time domain analysis. However, by checking the WAMIT file in the tutorial and the Hydro structure data in MATLAB, I found that it seems that only frequency domain analysis is needed in AQWA. I would like to confirm with you whether my conjecture is correct.
   2. It is known that setting conditions such as mooring and irregular wave for float structure needs to be carried out in the time-domain analysis module of AQWA. However, I found that there are corresponding setting statements for conditions such as irregular wave in WecsiminputFile. m file, so can I think of wecSim as time-domain analysis in AQWA? Here I want to confirm with you whether my idea is correct.
   3. I want to use AQWA to calculate and reproduce the hydrodynamic parameters of RM3, but RM3 is composed of two parts. as the figure 2 shows, I tried to set the two parts together, but when viewing the animation result, I found that the motion of the two parts did not move up and down as I thought, I am not sure whether there is something wrong with my simulation method. Please help me analyze the problem.
   
**Personal project context**
N/A

**Describe the type of conclusion or resolution you are looking for**
_I am still not clear about the hydrodynamic calculation of the device with two parts(just as RM3) . I hope you can tell me the specific method of using AQWA to calculate the RM3 coefficient. 
Since this part is an important part of my master thesis, I am looking forward to your reply！

**Additional context**

[image]
(figure 1)
[image]
(figure 2)

nathanmtom, 2022-08-19:
@sinclairwan Thank you for your question and we will do our best to respond, but please be aware that this issues board is focused on the use of WEC-Sim whereas your question appears to be predominantly focused on the use of AQWA. 

If you are attempting to reproduce the RM3 design within AQWA, we do have example files in the WEC-Sim repository: 

https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/AQWA/RM3

In response to your questions:

1. In WEC-Sim we use the Cummins formulation to complete simulations in the time domain and WEC-Sim converts frequency domain coefficients to their time domain representations as discussed in our WEC-Sim theory documentation.
2. Yes, WEC-Sim is a time domain solver and AQWA has its own time domain solver as well. 
3. Unfortunately in my initial statement to this issue, this issues board is for WEC-Sim usage and not AQWA. Therefore, I can only direct you to our example folder for AQWA runs to assist you. We cannot help you debug your input file generation to AQWA. 

I hope this helps point you in the right direction to troubleshooting the issues you are encountering. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sinclairwan, 2022-08-20:
@nathanmtom  Thank you for your answer! This has helped me a lot!

sinclairwan, 2022-08-22:
@nathanmtom 
Thank you for your patient answer！ I have checked the Lis file information of AQWA provided in EXAM folder, and found that there is no information such as mooring. AQWA will only set mooring in time domain analysis. May I assume that the AQWA input file you provided is only the result of frequency domain analysis?

nathanmtom, 2022-08-22:
@sinclairwan Yes, that is correct the AQWA files we provide are used to generate the frequency domain hydrodynamic coefficients which can be used in a frequency domain analysis. WEC-Sim will transfer the frequency domain coefficients to their time-domain representations to run a time domain simulation. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sinclairwan, 2022-08-23:
@nathanmtom  Thank you for your prompt reply! I will try to reproduce the frequency domain calculation results of RM3 with Aqwa. Do you have any suggestions for the parameter setting of Aqwa? Thank you!

nathanmtom, 2022-08-23:
@sinclairwan I'm glad my response helped answer your question. Since AQWA is outside of WEC-Sim, I do not have any suggestions or recommendations for AQWA parameter settings. Starting with any templates or examples AQWA has would be my only suggestion. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sinclairwan, 2022-08-24:
@nathanmtom Ok, thanks again for your reply. You have solved a large part of my problems. I will study the AQWA file in the example folder by myself to complete the simulation of my own project.