WEC-Sim/WEC-Sim #937: [Theory or Implementation] forceAddedMass

Hi-gone, 2022-08-24:
**Is your question request related to a problem? Please describe.**
I tried to fix the system pressure and find the best PTO force by changing the piston cylinder area . But I found that the forceTotal increases with the piston cylinder area, and the forceAddedMass seems abnormal at this time

**Describe the theory or implementation approach that you have a question**
In the RM3 example, I have set a pressure of constant magnitude in the opposite direction of the float movement
As shown in the figure below, I added a constant pressure of 20MPa to the PTO, and set the piston cylinder area to 2m^2, at which time the forceAddedMass sought became particularly large. And the abspower absorbed by the floating body has exceeded the wave power

**Personal project context**
'N/A' 

**Describe the type of conclusion or resolution you are looking for**
I think the floating body should not be able to move when a large PTO force is added, but the simulation results appear to be anomalous, and I would like to know the reason for the anomalous results
**Additional context**
_Add any other context or screenshots about the feature request here._
[image]
[image]

dav-og, 2022-08-24:
Hello, @Hi-gone, thank you for filling in our template.
Please can you upload a minimum working example so that we can reproduce your results?

Hi-gone, 2022-08-24:
@dav-og ，thanks for your reply，I've uploaded an example of a simple modification on RM3
[RM3.zip](https://github.com/WEC-Sim/WEC-Sim/files/9414367/RM3.zip)

dav-og, 2022-08-24:
Thanks @Hi-gone,

> I think the floating body should not be able to move when a large PTO force is added

Your PTO force does reduce the motion of the float a lot, but not completely. It doesn't seem anomalous to me.

However, the fAddedMass force is slightly larger than it should be:

[image]

This is because WEC-Sim sums the on-diagonal translational added mass terms and adds one value to the body's scalar mass term. Hence, for the RM3 float, the added mass in heave should be 1232.8*rho, but you actually have (92.7+92.7+1232.8)*rho:
[image]

If you're only considering heave in your simulations, you can try using simu.adjMassFactor to 'back out' the correct added mass value (see: https://github.com/WEC-Sim/WEC-Sim_Applications/blob/master/Free_Decay/1m/wecSimInputFile.m)

Other than that discrepancy, the model seems fine to me.

Hi-gone, 2022-08-25:
@dav-og thanks for your reply
I still have some doubts, when I increase the piston area by ten times(20m^2), the forceTotal also increases by ten times, obviously the wave power should have an upper limit. In addition, the average abspower of the float at this point is about 1.6e6kw, which exceeds the wave power.
[image]
Piston area set to 20
[image]
forceTotal also increases ten times
[image]
The average power is too large

dav-og, 2022-08-25:
Yes, I agree actually - I just commented out your PTO model and used our PTO model and have much more reasonable results:

[image]

[image]

Files attached:
[RM3_pto.zip](https://github.com/WEC-Sim/WEC-Sim/files/9423433/RM3_pto.zip)

I'm not sure where your model went wrong though - your block diagram seems reasonable 🤷‍♂️

Hi-gone, 2022-09-02:
Thanks @dav-og 
I made some modifications to the RM3_cHydraulic_PTO model to make the system pressure constant, and when I increased the piston cylinder area, this abnormal result appeared again.
[image]
[RM3_cHydraulic_PTO.zip](https://github.com/WEC-Sim/WEC-Sim/files/9475500/RM3_cHydraulic_PTO.zip)

So it seems that this is not a model problem, but maybe some bug in the calculation procedure of the addedmass force (because the addedmass force is more unusual compared to other forces in heave forces)
Can you provide some other method to estimate the optimal PTO force for a floating body？

jleonqu, 2022-09-27:
Hello @Hi-gone 
I was checking your model and I can have reasonable results when I am using the original value for the piston area, that is 0.0378 m^2. I think having an area of 1 m^2 is too much and you won't have good results from that. Probably any simulation will crash with such a big piston area.
I would suggest to find a reasonable range of values for the area, and then you can use the range as a constraint for your optimization function to estimate the optimal PTO force.

jleonqu, 2022-10-05:
Hello @Hi-gone 
Does this answer your question?

jleonqu, 2022-10-12:
I am closing this issue, but feel free to open a new one if you have more questions.

closed 2022-10-12 by jleonqu