WEC-Sim/WEC-Sim #939: [WEC-Sim Applications]There is a problem in the simulation using the H5 file written by myself

sinclairwan, 2022-08-26:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
$WEC-Sim\WEC-Sim_Applications-master\PTO-Sim\RM3TEST

**Describe the bug or issue**
     I used Aqwa to perform frequency domain analysis on RM3. The parameters set are the same as those in the example Aqwa file in ($WEC-Sim\examples\BEMIO\AQWA\RM3), and the LIS file and AH1 file are obtained. I hope to obtain the same hydrodynamic data as in the example and use it in the time domain analysis of wec-Sim. I replaced the tutorials\RM3\hydroData file with the AH1 file and LIS file that I had calculated myself, and I simulated the same result.
     But after I replace the AH1 file and the LIS file with the AQWA file in $VEC-sim \ VEC-sim_applications -master\ Pto-sim \RM3TEST\hydroData, However, wecsiMinputFile. m fails to run properly.

**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '$WEC-Sim\WEC-Sim_Applications-master\PTO-Sim\RM3TEST'_
2. _Click on 'wecSimInputFile.m'_
3. input ' wecSim '
4. _See error_
     错误使用 wecSim (第 45 行)
['RM3_cHydraulic_PTO/Global Reference Frame/Solver Configuration']: Error evaluating equations at time 0.0.  Stopping simulation.
There may be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by
tightening the error tolerances)
Caused by:
    错误使用 wecSim (第 45 行)
    ['RM3_cHydraulic_PTO/Translational PTO Actuation Force1/Translational PCC (Prismatic Joint)']: 'RM3_cHydraulic_PTO/Translational
    PTO Actuation Force1/Translational PCC (Prismatic Joint)' has a degenerate mass distribution on its follower side.

**Expected behavior**
After using the AH1 file and LIS file I provided, the output is as normal as the original example.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS:  Windows 10
 - [MATLAB/Simulink Release]   2021a
 - [WEC-Sim Release] v4.0

**Additional context**
I made my source files into a zip package, which you can find in $WEC-sim \ WEC-sim_applications-master \ PTO-Sim \RM3TEST, hydroData data files have been replaced.
[RM3TEST.zip](https://github.com/WEC-Sim/WEC-Sim/files/9432101/RM3TEST.zip)

nathanmtom, 2022-08-26:
@sinclairwan Thank you for your question and I will do my best to answer your question. 

Unfortunately, I cannot reproduce your error as when I copy the .h5 file generated from `$WEC-Sim\examples\BEMIO\AQWA\RM3` to the `$WEC-Sim_Applications\PTO-Sim\RM3\RM3_cHydraulic_PTO` folder I am able to run the simulation and there is no issue as you have shown above. 

I could not quite follow the procedure you listed above, but is your issue related to running with your own AQWA data and that is causing the error? The files you have included in RM3Test.zip appear to be from your own runs and does not include the files that are causing you errors (if I have followed your issue correctly).

If I have misunderstood your approach please clarify so I might be able continuing to help with your issue.

Cheers, 
Nathan 
National Renewable Energy Laboratory

sinclairwan, 2022-08-27:
@nathanmtom 
    Thank you for your generous help, which helped me a lot.
     After reading your answer, I'm sorry that you may have opened the wrong folder. The h5 file generated by myself is placed in the folder of $WEC-SIM \ WEC-Sim_Applications-Master \ PTO-Sim \ RM3TEST. For convenience, I packed my two AQWA files（TEST.LIS and TEST.AH1） and bemio.m into a compressed package, you can directly use them to generate h5 files ,and then use the model under $WEC-Sim_Applications\PTO-Sim\RM3\RM3_cHydraulic_PTO for simulation, then you will see the error.
     I am looking forward to your anwser!
Cheers,
Sinclairwan,

[AQWA source file.zip](https://github.com/WEC-Sim/WEC-Sim/files/9437157/AQWA.source.file.zip)

nathanmtom, 2022-08-29:
@sinclairwan Thank you for the clarification in the issue you are having. 

After using the AQWA files you attached in your previous comment I was able to reproduce your error. The issue seems to be occurring when attempting to run either `irregular` or `regularCIC` cases. The simulation did run with `regular` which means the issue seems to be coming from the infinite frequency added mass term or the convolution integral calculation. I did notice that if you look at our example RM3 AQWA case compared to your AQWA scripts that the infinite frequency added masses are quite different. Therefore, my suggestion is to see if you can fine tune your AQWA results and try to compare against our example AQWA case as a benchmark to try and determine what options to include in AQWA to get your simulation to run properly. 

Unfortunately, this is more of an AQWA issue rather than WEC-Sim and there is little support we can provide on how to update your AQWA runs. 

Cheers, 
Nathan
National Renewable Energy Laboratory

sinclairwan, 2022-08-31:
@nathanmtom 
According to your meaning, is there something wrong with the frequency setting. Could you please show me the screenshot of the position of the added mass under the infinite frequency in the LIS file, or where you think there might be an error,  so that I can easily modify it, thank you very much!

nathanmtom, 2022-08-31:
@sinclairwan

The AQWA file that we have as an example and your files do not directly output the infinite frequency added mass and instead we using the Fourier transform of the frequency dependent added mass coefficients per line 69 - 82 in `radiationIRF.m`. Therefore the issue of reproducing the infinite frequency added mass should not be related to your frequency setting. Rather this likely means that your added mass values are different than the RM3 example we provide which might be different because of the mesh used (or generated within AQWA) or other settings within AQWA when executing a run to calculate the frequency domain hydrodynamic coefficients.

Cheers, 
Nathan
National Renewable Energy Laboratory

sinclairwan, 2022-08-31:
@nathanmtom 
Thank you for your sincere guidance. Today I tried various methods to fix the problem I encountered, and finally found that it is necessary to change the GenerateinternalLid option to Yes during frequency domain analysis, and then the LIS file calculated again can be normal in wecSim run! My problem is now solved, I will close this question, thank you again!

closed 2022-08-31 by sinclairwan