WEC-Sim/WEC-Sim #943: [BUG]

hresing, 2022-09-02:
**Describe the bug**
Running BEMIO with either WAMIT .out files or AQWA .LIS and .AH1 files results in a hydro structure as desired, but with the ex_ prefix and fk_ prefix matrices being less than 6x6x(the number of frequencies relevant). In most cases it is 6x5, but sometimes is smaller.
I am most concerned with the ex_ma matrices, as that is the information relevant to what I am working on.

**WEC-Sim file(s)**
readAQWA, readWAMIT 
I would add examples of the .LIS and .AH1 files here, but github doesn't support those file types

**To Reproduce**
1. Open WEC-Sim
2. Open a folder with a bemio script and appropriate .AH1/.LIS or .out files
3. Run code
4. Open hydro structure
5. Check size of ex_ prefix matrices, most importantly the ex_ma one.

Running the BEMIO code via matlab GUI/command window

**Expected behavior**
I would expect the ex_ prefix matrices to be 6x6xthe number of frequencies, and indeed in the past that has been the size I have gotten.
Note that this switch to not 6x6 matrices did NOT occur when I pulled v5 of WEC-Sim, it has been happening before that as well.

**Screenshots**
n/a

**Workstation Information (this section MUST BE completed):**
 - OS:  Windows 10
 - MATLAB/Simulink Release 2022a
 - WEC-Sim Release v5.0 (pulled today)

**Additional context**
I have checked some of the AH1 and LIS files that result in this behavior, and they do have full 6x6x100 sets of data for the hydrodynamic parameters in them. (in other words, no missing columns or NaNs)

akeeste, 2022-09-08:
Hi @hresing 

These excitation force coefficients (``ex_XX, sc_XX, fk_XX``) within the BEMIO hydro struct are the correct size. Their dimension should be: [number of DOFs, number of wave headings, number of frequencies]. So this second dimension is not necessarily of size 6, but depends on number of wave headings. The [BEMIO hydro struct](https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#bemio-hydro-data-structure) section of the documentation details all information about these parameters. 

For example, if we run BEMIO for the `examples/BEMIO/AQWA/Cylinder/` example, we see that `ex_ma` has size 6x3x50. In the [corresponding LIS file](https://github.com/WEC-Sim/WEC-Sim/blob/master/examples/BEMIO/AQWA/Cylinder/cylinder.LIS), lines 8927-8977 show a sample of the 6 dofs, 3 headings, and 50 frequencies.

akeeste, 2022-10-05:
Hi @hresing 

I have not heard back so I will close this issue. Please reopen it and clarify if this has not answered your question. 

Adam