WEC-Sim/WEC-Sim #949: [BUG]The input parameter Settings of wecSimInput.m are incorrect

sinclairwan, 2022-09-18:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
When I set the waves.height and waves.period in the wecSimInput.m file, the error as shown in the figure appeared at runtime.

**WEC-Sim file(s)**
I modified the value of waves.period while keeping waves.height=2.5 unchanged, and found that when the value of waves.period is greater than or equal to 3.5, it can run normally. I doubt whether waves.period is set what precautions?
_Please upload relevant case files if possible to speed debugging._
[RM3TEST.zip](https://github.com/WEC-Sim/WEC-Sim/files/9593481/RM3TEST.zip)

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to 'wecSimInputFile.m'_
2. change the Value of "waves.period"
4. input 'wecSim' in command window_
5. _See error_

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
Hope to get a solution to this problem

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) 2021b
 - [WEC-Sim Release]v4.1

**Additional context**
_Add any other context about the problem here._

dforbush2, 2022-09-20:
Hi @sinclairwan 

I am looking into this. The first recommendation is to update to WEC-Sim v5.0 if possible. PTO-Sim was significantly overhauled in this release, so updating to the new release may help.

I was able to successfully run your input file with the WEC-Sim Applications PTO-Sim example: RM3_cHydraulic_PTO, on which your wecSimInputFile appears to be based. However, you are using your own BEM data: this appears to be the issue. 

You have some extremely large negative radiation damping values in surge and pitch at ~1.86 rad/s. When the wave excitation is near this frequency, the resulting run is likely to be unstable. Run plotBEMIO(hydro) at the end of your bemio.m script to see this. You will need to re-calculate your BEM results to stabilize your simulation. 

At a glance, I would recommend using the irregular frequency removal method of WAMIT (IRR = 1 in your *cfg file). 
 
Best,
Dom

dforbush2, 2022-09-28:
Hi @sinclairwan 

Did this solve your issue?

sinclairwan, 2022-09-29:
I am sorry that I have been a little busy these days and have not had time to reply your message. Next, I will update wecSim to the latest version according to your suggestion. Based on your suggestion, I would like to say that I used AQWA to get the BEMIO data, and I repeated the work after the LIS file content given in the example. I am not sure where the setting is wrong, can you give me some suggestions?

sinclairwan, 2022-09-29:
@dforbush2  Hi! I am sorry that I have been a little busy these days and have not had time to reply your message. Next, I will update wecSim to the latest version according to your suggestion. Based on your suggestion, I would like to say that I used AQWA to get the BEMIO data, and I repeated the work after the LIS file content given in the example. I am not sure where the setting is wrong, can you give me some suggestions?

dforbush2, 2022-09-30:
No worries.

I don't have any specific advice about AQWA, but generally radiation damping results should not be negative. Negative added mass is sometimes acceptable, but negative radiation damping is non-physical and unstable. If there is an irregular frequency removal option, I would use it. Resonance can cause numerical issues with the solver that these options can correct.

sinclairwan, 2022-09-30:
@dforbush2  I understand after hearing you say so, AQWA does have the option of eliminating the effect of irregular waves, I will try to change this option and recalculate, can you tell me what the final result is reasonable? I will use this as the basis for debugging.

dforbush2, 2022-09-30:
Try running ./examples/bemio/aqwa/rm3, and inspecting the plots that are produced (you may have to uncomment the line plotBEMIO(hydro) in bemio.m). The captions of these plots that will have some useful advice!

closed 2022-10-13 by dforbush2