WEC-Sim/WEC-Sim #951: MoorDyn example error - too many output arguments

t-osu, 2022-10-04:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_WEC-Sim application RM3 MoorDyn._

**Describe the bug or issue**
_I've used moordyn heavily in the past but now get an error running it, so I figured maybe I had not updated everything correctly. I have re-cloned wec-sim and moordyn, but have an error when I use moordyn for mooring.  I thought maybe it had to do with my model, but this occurs with the wec-sim applications RM3 moordyn case, too. _

**To Reproduce**
_For me, steps are the usual you take to run any wec-sim simulation, applied to the moordyn RM3 case_

**Expected behavior**
_To run successfully._

**Screenshots**
<
```
Error using wecSim
Error evaluating 'InitFcn' callback of SubSystem block 'RM3MoorDyn/MoorDyn1/MoorDyn/MoorDyn'.
Callback string is '% Initialize MoorDyn Lib (Windows:dll or OSX:dylib)
if ismac
    loadlibrary('Lines.dylib','MoorDyn.h');
elseif ispc
    loadlibrary('Lines.dll','MoorDyn.h');
elseif isunix
  loadlibrary('Lines.so','MoorDyn.h');
else
  disp('Cannot run MoorDyn in this platform');
end
calllib('Lines','LinesInit', mooring.location ,[0 0 0 0 0 0]);'
Caused by:
    Error using wecSim
    Too many output arguments.
```
>

**Workstation Information (this section MUST BE completed):**
 - OS:  Windows 10,
 - [MATLAB/Simulink r2022a
 - [WEC-Sim Release] v5.0.1

**Additional context**
_Thanks for any tips. Hopefully it is just something simple I am overlooking_

dav-og, 2022-10-05:
Hi @t-osu,

I just pulled the latest version of WEC-Sim and ran the same RM3MoorDyn example:
[image]

Its running fine for me:
[image]

>> I have re-cloned wec-sim and moordyn

Can I check which version of MoorDyn you cloned? It has to be v1. The repo I used is located here: https://github.com/WEC-Sim/MoorDyn

Cheers,
Dave

t-osu, 2022-10-05:
Morning Dave,
I just removed the moordyn i had and cloned from the link you sent to be sure of the version, but am still encountering the same error.

t-osu, 2022-10-05:
Dave,
I figured it out. Sort of similar to the other issue discussing this (#234). with the fbWecCntrl toolbox there is a source.m in their wafo folders and another startup.m. I removed everything for one startup.m.
all good know. thank you for the quick replies and glad it was an easy fix (figured it would be...)

closed 2022-10-05 by t-osu

dav-og, 2022-10-05:
Glad you figured it out!