WEC-Sim/WEC-Sim #952: [Theory or Implementation] Moments of inertia and Geometry orientation on import

Dadidal, 2022-10-04:
**Is your question request related to a problem? Please describe.**
N/A

**Describe the theory or implementation approach that you have a question**
I am want to understand how to correctly prepare the geometry and set Inertia values.

**Personal project context**
I try to model floating ship that consists of different parts. I want to use hydrodynamic body for floating part and non-hydrodynamic bodies for non-submerged parts. I created the geometry (.stl mesh files) for each of the ship part, but I am not clear how to set the Inertia in WEC-Sim input file.
The problem is that WEC-Sim requires input of 3x1 inertia matrix. Therefore these inertia terms are principal moments, right? Can WEC-Sim automatically calculate the direction of principal axes? This is important because geometry of ship parts in .stl file is oriented directly to the final position that should be on the ship. In such a case principal axes are not aligned with global axes during import (see Figure), but terms of Inertia in WEc-Sim called Ixx Iyy and Izz (not Ip1, Ip2, Ip3).

Should I rotate the parts in CAD so that the principal axes correspond to the global axes and only after that import the .stl file and manually change the orientation of the bodies already in WEC-Sim?

**Describe the type of conclusion or resolution you are looking for**
I would appreciate if you will explain how to prepare geometry or how to set inertia 3x1 matrix for the case if principal axes of the body in .stl file are not aligned along the global axes.

**Additional context**
[image]
[image]

akeeste, 2022-10-05:
Hi @Dadidal 

Thanks for reaching out. You are correct that this can be an issue and I will try to clarify.

WEC-Sim will not automatically calculate a body's principal axes. If there is no initial rotation, it is assumed that the coordinate system in ``*.stl`` file is the global coordinate system (horizontal x and y, vertical z). The ``body.inertia`` from the ``wecSimInputFile.m`` must be defined according to the coordinate system in the ``*.stl`` file. Ixx is about the file's x-axis, Iyy about the file's y-axis, etc. There are two ways you can do this:

**Option 1**
The first option is to export the CAD parts to ``*.stl`` in the global reference frame. So in the ``*.stl`` file, the axes are the global axes and the origin is the body's center of gravity. Set ``body.inertia`` in the ``wecSimInputFile.m`` to be the part's inertia about the global axes. This option makes the initialization in WEC-Sim very easy, but might make the inertia calculation and mesh creation more complex.

**Option 2**
The second option is to export the CAD parts to ``*.stl`` in their local reference frame. With the above blue boom as an example, this would be with z along the length of the boom. Set ``body.inertia`` in the ``wecSimInputFile.m`` to be the part's inertia about it's local axes. But then you must define an initial rotation in WEC-Sim using ``body.initDisp`` and ``body.setInitDisp()``. This is required so that the body ends up at the intended position. This option makes mesh creation and inertia calculation easier, but WEC-Sim initialization more complex.

> Should I rotate the parts in CAD so that the principal axes correspond to the global axes and only after that import the .stl file and manually change the orientation of the bodies already in WEC-Sim?

This corresponds to option 2 above. If you are already in CAD, I recommend using option 1 instead.

Please let me know if this is unclear and I can clarify further.
Adam

Dadidal, 2022-10-06:
@akeeste thank you for clarifying! Now I understand.

But I have a little question about Option 1 that you described. In such way where we define Inertia about global axes which are not aligned with local axes, the non-diagonal terms of Inertia matrix have non-zero values (Ixy, Ixz, Iyz). But WEC-Sim does not allow to define these terms, isn't it? Therefore Option 1 is not correct. Please correct me if I am not right.

akeeste, 2022-10-07:
@Dadidal Yes I understand your issue better now. That is correct, WEC-Sim does not currently allow the off-diagonal products of inertia. Simscape does allow this though and it seems straightforward to implement. In this case, especially with the yellow body the products of inertia may be significant. I added a [project board item](https://github.com/WEC-Sim/WEC-Sim/projects/58#card-86097325) for our team to address this. 

Before that feature is reviewed by our team, you could still use Option 1 by breaking the library links and adding the products of inertia manually here: `/Rigid Body/Hydrodynamic Body/Structure/Body Properties/Product of Inertia`. This seems straightforward, but I can't give further support until our development team has reviewed this in the source code.

For Option 2--
You can add another Rigid Transform block, but it is a bit redundant. WEC-Sim already uses these blocks within each body to implement the initial displacement, see ``/Rigid Body/Hydrodynamic Body/Structure/``. For that reason, I recommend using ``body.setInitDisp()``. This functions allows multiple rotations about a given point relative to the body's center of gravity. Based on the above figure, the green and blue bodies should only require one rotation, while the yellow body requires two. I can assist with setting up this function as needed.

Adam

Dadidal, 2022-10-08:
@akeeste Thank you for answer! Now I understand this options completely. What do you think about the usage of Simscape default rigid botdies in current situation to replace non-hydrodynamic bodies? This bodies allow to input 3x3 inertia tensor and it seems much easier than options described earlier.

akeeste, 2022-10-10:
@Dadidal On using default Simscape rigid bodies:
Yes you can do this and link to any other blocks or other Simscape libraries you want. Nothing in WEC-Sim should prevent default rigid bodies from working, though we can't provide debugging support for custom work. There are a couple things to note if you go this route:
- Gravity is off in the simulation. Must add a gravity force at the CoG
- To allow easy initialization in complex systems, each top-level model connection line is at the origin. PTOs are translated from the origin to their location, bodies are translated from the origin to their CoG, etc. You must include a rigid transform to translate from the origin to the custom bodies CoG.
- Output motion/forces will not be recorded by default
[image]
[image]

If you want to pursue this route and create custom bodies, I definitely recommend digging into the WEC-Sim body and PTO blocks to see what extra items are included there. 

On ``body.setInitDisp()``--
This function takes 3 inputs: the position of the CoG relative to the point of rotation, the list of rotation angles and their axes, and additional linear displacement. See this in detail with the commands ``help bodyClass.setInitDisp`` or ``doc bodyClass.setInitDisp``. Note that the principal axes you gave are not orthogonal, I'm guessing that either x or z should have a positive "0.014806" component. The [angle of the initial x axis](https://www.mathworks.com/matlabcentral/answers/101590-how-can-i-determine-the-angle-between-two-vectors-in-matlab) is 0.85 degrees about the y-axis. The command is then
 ``body.setInitDisp([0 0 0], [0 1 0 0.85], [0 0 0])`` and ``body.cg = [[0 0 -0.152]``

Please let me know if you have further questions
Adam

Dadidal, 2022-10-14:
@akeeste Thank you very much. I completly understood and solved this problem!

closed 2022-10-14 by Dadidal