WEC-Sim/WEC-Sim #953: how to link WEC-sim with external programs

JackyzhangHJ, 2022-10-06:
Hello, I have recently learned WEC-sim and would like to know how to link WEC-sim with external programs. I would like to know how to link WEC-sim with external dynamic link libraries (.dll files). In other words, I want to build the equations of motion of the upper structure of floating body through the dynamic link library, and I need to call the motion response of the floating body obtained by WEC-sim in real time. At the same time, the dynamic link library will output the load at the location where the upper structure is connected to the floating body as an external load to WEC-sim. This creates a data exchange to achieve full coupling. Can this idea be realized?
In a sense, moordyn seems to be formed in this way. But I didn't find a specific way to implement it with the code though.
Best wishes.

akeeste, 2022-10-10:
Hi @JackyzhangHJ 

We require users to use a template when creating an issue. This helps our development team get needed information up front and answer questions more efficiently. Can you please modify your issue to use a template? (or create a new issue). I suggest the implementation or feature request templates for your question.

[image]
[image]
[image]

Thanks
Adam

JackyzhangHJ, 2022-10-11:
Hi, @akeeste 
   According to your comments, I have add a new issue in #954 by the template 'Theory implementation clarfication'. And I supply some my research progress in it. Thanks!

closed 2022-10-11 by dav-og