WEC-Sim/WEC-Sim #956: [Theory or Implementation] How to link WEC-sim with external program (for example, a .dll file)?

JackyzhangHJ, 2022-10-11:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
_I have read the user's guide of WEC-sim. I found that it does not describe in detail how WEC-sim calls external programs._

**Describe the theory or implementation approach that you have a question**
_In the User manual of WEC-sim, I found some contents about how to develop WEC-sim in the section 'Code Structure'. It just have two ways to call external program, recompile the code or use other Simulink/Simscape blocks currently, right?_

**Personal project context**
_Hello, I have recently learned WEC-sim and would like to know how to link WEC-sim with external programs. I would like to know how to link WEC-sim with external dynamic link libraries (.dll files). In other words, I want to build the equations of motion of the upper structure of floating body through the dynamic link library, and I need to call the motion response of the floating body obtained by WEC-sim in real time. At the same time, the dynamic link library will output the load at the location where the upper structure is connected to the floating body as an external load to WEC-sim. This creates a data exchange to achieve full coupling. Can this idea be realized?
Actually, I want to combine FAST (an open-source software to caculate the motion and dynamic response of wind turbine) and WEC-sim to research the system of combine wind and wave. I have recomplie the FAST program to a .dll file. So I want to know how to link WEC-sim with an external program.
I found there are some issues about wind-wave energy system in the WEC-sim issue page. There is no way to connect FAST and WEC-sim directly. I recompile FAST to a program caculate response of wind turbine only. So the wind turbine load is seem as the external load of WEC-sim. 
In a sense, moordyn seems to be formed in this way. But I didn't find a specific way to implement it with the code though._

**Describe the type of conclusion or resolution you are looking for**
_I still don't know if my idea how to achieve. I would like to point out in the documentation or code where I can get additional information. Or if you can recommend relevant articles for reference? Thanks._

**Additional context**
_Add any other context or screenshots about the feature request here._

dav-og, 2022-10-19:
Hi @JackyzhangHJ, I think you should just be able to use the same structure/approach that we use for the MoorDyn block - which as you can see from my screenshot is in 'WECSim_Lib_Moorings'. You could have block next to that for your FAST coupling...I can't see what you've tried so far (in terms of structure) - where is your 'FAST' block located?
Cheers,
Dave

dav-og, 2022-11-07:
Hi @JackyzhangHJ, I will close this for now as it seems to be an ongoing development effort - feel free to reopen or open a new issue if you have any further questions though.
Cheers,
Dave

closed 2022-11-07 by dav-og