WEC-Sim/WEC-Sim #959: [Theory or Implementation] Hydrodynamic coefficients recalculation

Dadidal, 2022-10-14:
**Is your question request related to a problem? Please describe.**
Good afternoon! I don't understand if the hydrodynamic coefficients (added mass, radiation damping, hydrostatic stiffness) will be consistent with the geometry if I change the position of non-submerged parts (and correspondingly CoG and Moments of Inertia).

**Describe the theory or implementation approach that you have a question**
N/A

**Personal project context**
I try to make a model which consists of one hydodynamic body and several non-hydrodynamic bodies (https://github.com/WEC-Sim/WEC-Sim/issues/859). I calculated hydrodata (A, B, KHs) using Capytaine. In Capytaine for calculation I used the geometry, mass and inertia of only submerged part (hull). Then I added topside parts as nonhydrodynamic bodies in WEC-Sim.

Recently I observed that Capytaine results are dependent on geometry that is used during calculation. For example hydrostatic stiffness matrix (KHs) is changing depending on whether I use the full mesh of the floating part or only the part of the mesh under water. Then I made a model in AQWA and got the completely different matrix of KHs. It seems that AQWA accounts the mass and inertia of all point masses (under and above water level). Also KHs in AQWA is changing when I move one of topside (non-submerged) parts of the model.

**Describe the type of conclusion or resolution you are looking for**
Considering above results I want to clarify following questions:
1. Are the hydrodynamic coefficients (A, B, Khs) are dependent on the position (CoG) of the each part? 
2. Can WEC-Sim autimatically change KHs matrix based on position and orientation of nonhydrodynamic bodies? Is it the option of nonlinear calculation of buoyancy forces?
3. If coeffitients are dependent on CoG position what CoG, Mass and Inertia should I use in Capytaine if I want to calculate only Submerge part (hull) - Hull parameters only or parameters related to full model?

**Additional context**
AQWA model:
[image]

Capytaine model:
[image]

Hydrostatic stifness matrix from Capytaine (submerged part only):
0 0 0 0 0 0
0 0 0 0 0 0
0 0 8447.126 0.0104 -471.935 0
0 0 0.0104 187.677 -0.0083 -34.404
0 0 -471.935 -0.0083 1799.957 -0.0002
0 0 0 0 0 0

Hydrostatic stiffness from AQWA (full model):
0 0 0 0 0 0
0 0 0 0 0 0
0 0 8442.998 0.1607 1.986 0
0 0 9.21 2.675 0.0018 ?
0 0 113.773 0.0018 30.360 ?
0 0 0 0 0 0

Hydrostatic stiffness from AQWA after moving one of non-submerged parts:
0 0 0 0 0 0
0 0 0 0 0 0
0 0 8442.961 -0.087 3.441 0
0 0 -4.988 2.674 -0.0024 ?
0 0 197.135 -0.0024 30.414 ?
0 0 0 0 0 0

dav-og, 2022-10-19:
Hi @Dadidal,

>> 1. Are the hydrodynamic coefficients (A, B, Khs) are dependent on the position (CoG) of the each part?

The hydrodynamic coefficients are calculated about the 'hydrodynamic' body's CoG - for its mesh in equilibrium. These coefficients are valid for small amplitude motion about the body's CoG. However, the motion of your crane arm and boom could influence the hull's equilibrium position and its wetted surface - leading to the hull's CoG resting in a different position to where you computed your hydrodynamic coefficients. Its up to you to determine if that invalidates the small amplitude motion assumption or not (it might be fine). For excitation and hydrostatics you could use our [nonlinear options](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#nonlinear-buoyancy-and-froude-krylov-excitation), to account for the wetted surface of the body at each time step (this should answer your Q2).

>> 3. If coeffitients are dependent on CoG position what CoG, Mass and Inertia should I use in Capytaine if I want to calculate only Submerge part (hull) - Hull parameters only or parameters related to full model?

You shouldn't need to define mass and inertia for Capytaine (only CoG). Since the hydrodynamic coefficients you need are for the hull, you should compute them about the hull's CoG. As discussed above, if the motion of the other bodies causes large changes in the hull's equilibrium position, you could potentially invalidate the hydrodynamic coefficients - but nonlinear hydrostatics and excitation could help.

In a slightly related problem, we have simulated large yaw displacements in the past - and interpolated between hydrodynamic coefficients as the body deviated from its initial equilibrium position. Please see [this publication](https://tethys-engineering.pnnl.gov/sites/default/files/publications/Forbushetal2020.pdf) for how we did that previously.

Cheers,
Dave

Dadidal, 2022-10-21:
@dav-og Thank you very much for your reponse. In conlusion I want to confirm the following facts. Please correct me if I am not right:
1. It can be said that nonlinear options will keep Khs matrix valid for instantaneous position of crane hull and its parts during simulation
2. Added mass (A) and radiation damping (B) coefficients are not dependent on position of upper parts. Also they will be valid if I calculate them in Capytaine for final equilibrium position (draft of full model in WEC-Sim with non-hydrodynami bodies).

If so, I am clear with this issue.

dav-og, 2022-11-02:
I'll close out for now - feel free to re-open if you need further clarification 👍
Cheers,
Dave

closed 2022-11-02 by dav-og