WEC-Sim/WEC-Sim #963: Unrecognized readCAPYTAINE

hfpim, 2022-11-07:
I'm trying to generate a .h5 file from a .nc file but I keep getting the following error:
*Unrecognized function or variable 'readCAPYTAINE'.*

I thought it was some wrongdoing of my part so I tried multiple WEC-Sim examples but I keep getting the same error.
How can I fix that?

dav-og, 2022-11-07:
hello @hfpim,

Please can you fill in one of our issues templates to help us figure this out?:

If you hit 'New issue' you should see the following options:
[image]

(I suggest filling in the first one)

cheers,
Dave

closed 2022-11-07 by dav-og

hfpim, 2022-11-07:
That option didn't show up before. Did it work now?

dav-og, 2022-11-07:
I'm not sure whats changed but if you open up a new issue & hit bug report then you should see the following template:
[image]
if you fill in these fields it can help us to advise you on your issue.