WEC-Sim/WEC-Sim #964: [BUG] Unrecognized readCAPYTAINE

hfpim, 2022-11-07:
I'm trying to generate a .h5 file from a .nc file but I keep getting the following error:
_Unrecognized function or variable 'readCAPYTAINE'.

I thought it was some wrongdoing of my part so I tried multiple WEC-Sim examples but I keep getting the same error. I tried sphere and cylinder examples without success
How can I fix that?

**Here is the code**

```
hydro = struct();
hydro = readCAPYTAINE(hydro,'cylinder_full.nc');
hydro = radiationIRF(hydro,15,[],[],[],[]);
hydro = radiationIRFSS(hydro,[],[]);
hydro = excitationIRF(hydro,15,[],[],[],[]);
writeBEMIOH5(hydro)
plotBEMIO(hydro)
```

The path to folder is: 
_\...\WEC-Sim-master\examples\BEMIO\CAPYTAINE\cylinder_
To run it, I just click "Run" on Matlab.

[image]

**Workstation Information:**
 OS: Windows 10
 [MATLAB/Simulink Release]: R2021b

dav-og, 2022-11-07:
Hi @hfpim thanks for the additional information,

I just did a fresh install of WEC-Sim-master and ran the Capytaine cylinder example...but it worked fine for me, so I couldn't reproduce your error:

[image]

Can I just check if you ran the 'addWecSimSource.m' script? (to install WEC-Sim and add it to your path)

(Full installation instructions available here: https://wec-sim.github.io/WEC-Sim/master/user/getting_started.html#install-wec-sim)

dav-og, 2022-11-16:
Hi @hfpim, are you still having a problem with the readCAPYTAINE() function or can we close out this issue?

hfpim, 2022-11-20:
I figured it out, you can close this issue, thank you!

closed 2022-11-21 by nathanmtom