WEC-Sim/WEC-Sim #965: [BUG]

abd0-polimi, 2022-11-16:
**Describe the bug**
I am new to wecSim and Simulink in general. I tried to do both provided tutorials in the wecSim documentation (RM3 and OSWEC). Although I followed the documentation instructions precisely, I got a problem with the Simulink file when running WEC-Sim. The problem is shown in the screenshot and my files are uploaded in a zip file. This problem does not happen when I replace the Simulink file that I wrote with that provided in the "examples" folder of wecSim source code. 

**WEC-Sim file(s)**
[RM3.zip](https://github.com/WEC-Sim/WEC-Sim/files/10022019/RM3.zip)

**To Reproduce**
'N/A'

**Expected behavior**
It should run without any problem.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB Release](https://www.mathworks.com/help/matlab/release-notes.html) Version 9.11 (R2021b)
 - [Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) Version 10.4 (R2021b)
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v5.0.1

**Additional context**
'N/A'

dav-og, 2022-11-16:
Hello @abd0-polimi, thank you for filling in the template.

The tutorials included with WEC-Sim do not work out of the box - there are a few steps missing that the user needs to figure out. Please check out the tutorials section of our documentation: https://wec-sim.github.io/WEC-Sim/master/user/tutorials.html#

The models included in the [Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) folder should work out of the box, as should the models on the [WEC-Sim Applications](https://github.com/WEC-Sim/WEC-Sim_Applications) repo

abd0-polimi, 2022-11-16:
Hi @dav-og, thanks for your reply. 

I have followed all the steps written in the documentation:  https://wec-sim.github.io/WEC-Sim/master/user/tutorials.html#
and I got the error shown in the warning. you can also check out the simulink file in the uploaded zip folder.

abd0-polimi, 2022-11-16:
Thank you for your help, But:
I changed the block to a translational PTO but I still got the same problem. 
[image]
[image]

dav-og, 2022-11-16:
it looks like you may have grabbed the 'Translational' block from the constraints library, rather than the 'Translational PTO' block from the PTO library:
[image]

abd0-polimi, 2022-11-16:
Hey, @dav-og it is working fine now.
Thank you so much!! what a rookie mistake from me :sweat_smile:

dav-og, 2022-11-16:
No problem :)

closed 2022-11-16 by dav-og