WEC-Sim/WEC-Sim #966: [WEC-Sim Applications]

accient, 2022-11-19:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_N/A_

**Describe the bug or issue**
Hello, I’m trying to repeat the verification in this essay (Coupled Mooring Analyses for the WEC-Sim Wave Energy Converter Design Tool : https://asmedigitalcollection.asme.org/OMAE/proceedings/OMAE2016/49972/V006T09A023/281290). 
[image]

But when I ran my model, I got an error in calling moordyn:
[image]

I had read the user guide of moordyn (4.2. Diagnosing Problems), and try to solve this problem as the document advised. But that doesn’t work. I'd appreciate it if anyone could help.
**To Reproduce**
The whole project was uploaded to my github:
https://github.com/accient/mooring_verification/tree/main/WEC-Sim

**Expected behavior**
_N/A._

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [ 2021b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v4.4]
 - [Moordyn Release](https://github.com/WEC-Sim/MoorDyn) [V1.01.02]

accient, 2022-12-01:
Hello @dav-og,

Thanks for your help. I reduce the dtM and threshIC as you say. That really works. 

But with this tiny dt, the model ran at a very slow speed. It seems that WEC-Sim coupled with mooordyn can't well handle these small WEC problems. Cause I saw the similar issue at https://github.com/WEC-Sim/WEC-Sim/issues/569.

Maybe I should try to scale the model to full scale. Whatever, thanks again for your help!

Best wishes.

dav-og, 2022-12-01:
Hi @accient, glad its working. I just used very small values to get the model working - I'm sure you could play around with these to find a better trade-off with speed...it may also be worth checking with Yi-Hsiang and Matt about how to find a good combination of dtM and threshIC values (and to check which values were used for the model in the paper).

Or, as you said - scaling the model up to full scale could also help.

Cheers! I'll close this out.
Dave

closed 2022-12-01 by dav-og