WEC-Sim/WEC-Sim #969: [WEC-Sim Applications]

wpoulter, 2022-12-07:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._
I am trying to use Capytaine to generate hydrodynamic data for a sphere for BEMIO so I can convert it to a .h5 file for use in WECsim. 

**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._
I have tried using my own script to create a .nc file which when used with the readCAPYTAINE function in MATLAB returns the following error messages.

Warning: Hydrostatics data not included in Capytaine output. Value of zero assigned to cg, cb, Vo. 
> In readCAPYTAINE (line 191)
In read_capytaine_output (line 5) 
Warning: No KH.dat files found. Hydrostatics can't be computed. 
> In readCAPYTAINE (line 236)
In read_capytaine_output (line 5) 
Error using  / 
Integers can only be combined with integers of the same class, or scalar doubles.

Error in normalizeBEM (line 59)
    hydro(F).Khs = hydro(F).Khs/(hydro(F).g*hydro(F).rho);

Error in readCAPYTAINE (line 434)
hydro = normalizeBEM(hydro);  % Normalize the data according the WAMIT convention

Error in read_capytaine_output (line 5)
hydro = readCAPYTAINE(hydro, "C:\simple_dataset.nc");

I have also tried modifying the code using the call_capytaine method included in the BEMIO examples with my own meshes. I have tried using an .stl exported from SOLIDWORKS and I have tried converting the .stl file to a .gdf using meshmagick. Both give the same output.

        --> the mesh does not seem watertight although marked as closed...
Clipping C:\Users\...\WEC-Sim\examples\BEMIO\CAPYTAINE\my sphere\sphere.stl_mesh by Plane(normal=[0. 0. 1.], point=[0. 0. 0.]): all vertices are removed.

-------------------------------
Calling Capytaine BEM solver...
-------------------------------
mesh = ('C:\\Users\\...\\WEC-Sim\\examples\\BEMIO\\CAPYTAINE\\my sphere\\sphere.stl',)
w range = 0.020 - 8.400 rad/s
dw = 4.190 rad/s
no of headings = 1
no of radiation & diffraction problems = 21
-------------------------------

Process Process-1:
Traceback (most recent call last):
  File "C:\Users\...\Miniconda3\lib\multiprocessing\process.py", line 315, in _bootstrap
    self.run()
  File "C:\Users\...\Miniconda3\lib\multiprocessing\process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\...\WEC-Sim\examples\BEMIO\CAPYTAINE\call_capytaine.py", line 331, in capy_solver
    capyData = solver.fill_dataset(problems, [combo], hydrostatics=False)
  File "C:\Users\...\Miniconda3\lib\site-packages\capytaine\bem\solver.py", line 161, in fill_dataset
    problems = problems_from_dataset(dataset, bodies)
  File "C:\Users\...\Miniconda3\lib\site-packages\capytaine\io\xarray.py", line 102, in problems_from_dataset
    DiffractionProblem(body=body_range[body_name], omega=omega,
  File "C:\Users\...\Miniconda3\lib\site-packages\capytaine\bem\problems_and_results.py", line 239, in __init__
    super().__init__(body=body, free_surface=free_surface, sea_bottom=sea_bottom,
  File "C:\Users\...\Miniconda3\lib\site-packages\capytaine\bem\problems_and_results.py", line 63, in __init__
    self._check_data()
  File "C:\Users\...\Miniconda3\lib\site-packages\capytaine\bem\problems_and_results.py", line 90, in _check_data
    raise ValueError(f"The mesh of the body {self.body.name} is empty.")
ValueError: The mesh of the body sphere_cpt is empty.
Traceback (most recent call last):
  File "C:\Users\...\Miniconda3\lib\site-packages\xarray\core\concat.py", line 221, in concat
    first_obj, objs = utils.peek_at(objs)
  File "C:\Users\...\Miniconda3\lib\site-packages\xarray\core\utils.py", line 184, in peek_at
    peek = next(gen)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\...\WEC-Sim\examples\BEMIO\CAPYTAINE\my sphere\simple_BEM_case_with_call_cpt.py", line 33, in <module>
    cc.call_capy(meshFName = bem_file,
  File "C:\Users\...\WEC-Sim\examples\BEMIO\CAPYTAINE\call_capytaine.py", line 273, in call_capy
    capyData = read_netcdfs(ncFName, dim='omega')
  File "C:\Users\...\WEC-Sim\examples\BEMIO\CAPYTAINE\call_capytaine.py", line 306, in read_netcdfs
    combined = xr.concat(datasets, dim)
  File "C:\Users\...\Miniconda3\lib\site-packages\xarray\core\concat.py", line 223, in concat
    raise ValueError("must supply at least one object to concatenate")
ValueError: must supply at least one object to concatenate

**To Reproduce**
_Steps to reproduce the behavior:_

My Python code using Capytaine 

from capytaine import *
import logging
#logging.basicConfig(level=logging.INFO)

#Creating mesh for sphere
resolution = 16
sphere = Sphere(radius=1, center=(0, 0, 0), ntheta = resolution, nphi = resolution, name="my buoy") #ntheta and nphi control the resolution of the mesh

#Giving object dofs
sphere.add_all_rigid_body_dofs() #method to all dofs to an object

#Set centres
sphere.center_of_mass = (0, 0, 0)
sphere.rotation_center = (0, 0, 0)

#Crops part of mesh that is not immersed in water (Seems to be the immersed portion when statically sitting in water)
#Must do this to compute a radiation problem
sphere.keep_immersed_part()

#Compute all hydrostatic parameters (Hydrostatic stiffness and inertia matrix)
# sphere.compute_hydrostatic_stiffness()
# sphere.compute_rigid_body_inertia()
# sphere.hydrostatics = sphere.compute_hydrostatics()
sphere.inertia_matrix = sphere.compute_rigid_body_inertia()
sphere.hydrostatic_stiffness = sphere.compute_hydrostatic_stiffness()
#print(sphere.hydrostatics)

#Defining a linear potential flow problem
from numpy import infty
#Infinity number from MATLAB, used for depth
# NB Values in problems should not conflict, only radiating_dof and wave direction will be different
problem = RadiationProblem(body=sphere, radiating_dof="Heave", omega=1.0, sea_bottom=-infty, g=9.81, rho=1000)
problem2 = DiffractionProblem(body=sphere, omega=1.0, sea_bottom=-infty, g=9.81, rho=1000)

#Solving a linear potential flow problem
solver = BEMSolver()
result = solver.solve(problem)
result2 = solver.solve(problem2)

from capytaine import assemble_dataset
dataset = assemble_dataset([result, result2], hydrostatics = True)
print("*******************************")
print(dataset)
#Workaround to write complex values to NetCDF file
from capytaine.io.xarray import separate_complex_values
separate_complex_values(dataset).to_netcdf("simple_dataset.nc")

Modified included Python code using call_capytaine

# setup environment
import os
os.environ["OMP_NUM_THREADS"] = "1" 

import numpy as np
import sys

# Add directory with the call_capytaine.py file to the system path.
currentdir = os.path.dirname(os.getcwd())
sys.path.append(currentdir) #Filename used to be currentdir
import call_capytaine as cc

# Define sphere parameters ---------------------------------------------------#
bem_file = (('C:\\sphere.stl'),) # mesh file, .dat nemoh, .gdf wamit
bem_cg = ((0,0,0),)                                   # center of gravity
bem_name = ('sphere_cpt',)                               # body name

bem_w = np.linspace(0.02, 8.4, 3)                      # wave frequencies
bem_headings = np.linspace(0,0,1)                        # wave heading
bem_depth = 50.0                                         # water depth

bem_ncFile = os.getcwd() + os.path.sep + 'sphere.nc'     # path for output .nc file
# ----------------------------------------------------------------------------#

# Run Capytaine
if __name__ == '__main__':
    cc.call_capy(meshFName = bem_file,
                 wCapy     = bem_w,
                 CoG       = bem_cg,
                 headings  = bem_headings,
                 ncFName   = bem_ncFile,
                 body_name = bem_name,
                 depth     = bem_depth,
                 density   = 1000.0)

**Expected behavior**
_A clear and concise description of what you expected to happen._
When I am using my own Python script I was expecting BEMIO to recognise the hydrodynamic parameters in the simple_dataset.nc file. I am also unsure why it is giving an error using /.
When I am using the included Python script with call_capytaine I am unsure why my meshes are causing errors in call_capytaine, all the vertices are getting deleted even though the sphere is centred in the .stl file and has CoG at (0,0,0).

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release] R2022b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) I am unsure how to check the version but I downloaded it using Git two days ago so I think it should be the latest stable version. I saw a recently approved PR was implemented in one of the functions I had so I think it is the most recent release.

**Additional context**
_Add any other context about the problem here._

wpoulter, 2022-12-07:
Sorry, the output looks incorrect as Github has used it as formatting.

wpoulter, 2022-12-07:
Hi @dav-og,

I am now able to use the readCAYTAINE, and calculate the radiationIRF, radiationIRFSS, and excitationIRF parameters. I implemented the call_capytaine functions for writing .dat files in my code as you suggested and these are now being read in MATLAB. The other issue I had was already documented (https://github.com/WEC-Sim/WEC-Sim/issues/850) and required me to add a decimal point after the density value. 
However, I am not able to write an .h5 file. I get the following error messages in MATLAB when using writeBEMIOH5

Error using matlab.io.internal.vfs.validators.hasIriPrefix
Argument must be a text scalar.

Error in H5F.create (line 78)
is_remote = matlab.io.internal.vfs.validators.hasIriPrefix(varargin{1});

Error in writeBEMIOH5 (line 18)
fileID = H5F.create(filename,'H5F_ACC_TRUNC','H5P_DEFAULT','H5P_DEFAULT');

Error in read_capytaine_output (line 14)
writeBEMIOH5(hydro)

This must be an issue with my output from Capytaine, do you know what would cause this? I have attached my python script for generating the required .nc and .dat files and my MATLAB script for writing an .h5 file. [BEMIO files.zip](https://github.com/WEC-Sim/WEC-Sim/files/10180244/BEMIO.files.zip) 
Thanks for your help,

Will

wpoulter, 2022-12-07:
Hi @dav-og,

I was also wondering if you have any suggestions for free cad software that can generate a usable .stl file?
Cheers,

Will

wpoulter, 2022-12-08:
I found my issue described above. I had the following in my code:

hydro = readCAPYTAINE(hydro, "mysphere.nc");

This caused this error:

Error using matlab.io.internal.vfs.validators.hasIriPrefix
Argument must be a text scalar.

This error occurred because of the quotation marks. Using apostrophes instead of quotation fixed the error, the following code ran as expected:

hydro = readCAPYTAINE(hydro, 'mysphere.nc');

I was not aware there was a difference in using quotation marks and apostrophes in MATLAB syntax but there clearly is.

wpoulter, 2022-12-08:
https://au.mathworks.com/matlabcentral/answers/485385-difference-between-single-quote-vs-double-quote

closed 2022-12-08 by wpoulter

dav-og, 2022-12-08:
Cheers @wpoulter, glad you figured it out!

I was actually looking over the .py script you shared and cleaned a few things up - e.g. generally its bad practice to have things like `from xyz import *` - this can make it difficult to determine where a particular function or variable is coming from and potentially cause naming conflicts.

Sharing my updated script as a .txt file:

[simple_BEM_case-DO.txt](https://github.com/WEC-Sim/WEC-Sim/files/10189110/simple_BEM_case-DO.txt)

Cheers,
Dave